sharpie7/circuitjs1 #638: Bug-  p-mos transistors and a current source

pilliam-wugsley, 2021-08-20:
This is a super website that I am using to teach analogue circuit design and my pupil has come across a bug.

When driving a current p-mos current mirror from a 10mA current source the currents and voltages appear to be incorrect. The circuit is not working at all.  We have drawn this on two separate windows 10 machines in firefox, chrome and edge.

This is the circuit:  

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKADNwVsURMexshKnxDZs4aEhiQULAG4hiAFiXghilSCV4oupbV1UYCFgHdO3Xhn5ce3I2fCCqSsISfqUeHZEdhbomgWPK7uvub+lqEeLm5Qfs4U6sqqCOrhGqnJmm4O5ilaOgLq2vERiaXFwmnxAOaZSVQFhDUOHGBK1lb8CILdouJgklCwshGdPCIdXfZl4BOBVJF2QRlgve4oNet9pWsbhTGH+31eRQdpeeAX6lWN8QCWR9F3l4awrEA

[image]

pfalstad, 2021-08-20:
yes this is a bug.  the simulator doesn't see a current path for the current source, so it's not enabled.

if you put a 1Meg resistor in parallel with the current source, it will work.

pilliam-wugsley, 2021-08-20:
Ah yes that works perfectly.  Just a suggestion: I wonder if a 1G ohm resistor could be put in parallel to the current source inside the current source

pilliam-wugsley, 2021-08-20:
Solution for those who want to know:  https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKADNwVsURMexshKnxDZs4aEhiQULAG4hiAFiXghilSCV4oupbV1UYCFgHdO3Xhn5ce3I2Yt20TrWEJRHYW248D1Su6e5v606sqqCOqQjhEU4ZruDuZx2uCCVGkxIRlaOqG8UZ4A5hqRCaqERQ4cYErWVvwIgo2i4mCSULCyIfU8InUN9sHgfaIu3pbD2eDNHihFYHN5I0staQVZXssoePnLUQ4AlukBQQWHhrCsOWd+udh4qjMFj6oFuzovD0+jRW+eABOfw8ALqRUuVDA8BhcC8uUu4I8kK8Sgh6jWyOiLCAA

[image]

pfalstad, 2021-08-20:
Re-opening this because I still need to fix it...

pilliam-wugsley, 2021-08-20:
sorry - my bad

pfalstad, 2021-08-22:
fixed now

closed 2021-08-22 by pfalstad