sharpie7/circuitjs1 #642: Feature Request: Op-amp Wein Bridge Oscillator sample circuit.

marcdraco, 2021-09-01:
Possible example circuit? This is a 1.6Khz sine oscillator without an AGC so it's as twitchy as a twitchy thing that's wired on far too much coffee.

Not at all exciting but I thought it was an interesting example. The gain is adjustable to demonstrate just how finicky this thing is to balance. It's going to be interesting to build one of these for real (sans Bill Hewlett's tungsten bulb as here) and see if it's possible to get it to stabilize. I expect it will fail quite spectacularly with a real amp.

```
$ 1 0.000015625 26.59566520631553 47 5 50 5e-11
a 80 96 368 96 8 15 -15 1000000 4.064786910362269 4.064910108737269 100000
r 80 208 224 208 0 1000
c 256 208 368 208 0 1.0000000000000001e-7 -0.7936139166191598 0.001
w 224 208 256 208 0
w 368 208 368 96 0
w 80 112 80 208 0
r 80 208 80 384 0 1000
w 80 208 -16 208 0
c -16 208 -16 384 0 1.0000000000000001e-7 4.064910108737269 0.001
w -16 384 80 384 0
w 80 384 80 432 0
g 80 432 80 464 0 0
w 80 80 -112 80 0
r -112 80 -112 240 0 49240
g -112 240 -112 320 0 0
r 0 0 368 0 0 100000
w 368 0 368 96 0
w 0 0 -112 0 0
w -112 0 -112 80 0
211 368 96 432 96 0 1 8000 1
o 4 2 0 x8301a 0.0001 0.0001 0 2 5 0 4 3 0.025 0
38 13 0 39000 55000 Resistance

```

pfalstad, 2021-09-01:
Thanks..  But we have one already, under Circuits->Misc Devices->Light Bulb.  A bit hard to find, I admit.

marcdraco, 2021-09-01:
Ah damn, I missed that. Thanks Paul. ;)

I'm having far too much fun wit this. ;)

closed 2021-09-01 by marcdraco