sharpie7/circuitjs1 #646: Bug- False wire loop detection

sambthompson, 2021-09-05:
If I have two triple (or more) throw switches in a circuit, even if not connected to make a loop, I still get "wire loop detected". [A simple example](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKAGURs1PIAWEXmEI9+VCADMAhgBsAznXCd2nbrzgChA9WJBS5CiNhYB3FVRQJhXKrlEmNVvgN54RUe4OEo8r3i5DerpAemmpUfr7aIcKezr6awaYRDnFaVImptqmB7kn+OS7mPrmpsYUpwWB4EOU5lkVBItDElmAIxCgoxJAWkEhg6mDcMKLQvCxVNXjhmvUVPGKDw2NQYywA5hSEDVs2w+6bc7FH-uksAEpmAZYBGCjXwula1E8wCPbWbp9h7kA).

I know there's a common issue where two switches are arranged together alternately control a single output creates a loop - this is not that case - each switch is controlling its own output (lamps used in sample).

pfalstad, 2021-09-05:
Fixed.  It seems we didn’t handle the case where two of the switch terminals were connected by wires.

closed 2021-09-05 by pfalstad

sambthompson, 2021-09-05:
Thanks, Paul - fantastic response and application!