sharpie7/circuitjs1 #647: Feature request: oscilloscope octopus

marcdraco, 2021-09-06:
I don't know if you've done this, there are many many versions of it but the simplest ones require nothing more than an AC source and a current limiting resistor.

The "device under test" is placed to complete the circuit and the voltages are monitored in X-Y producing simple Lissajous figures describing how the component works. 

Although not that useful in a simulator, it shows how various devices behave in a circuit like this and I've found it exceptionally useful to help develop something for my own scopes (and a standalone project for Arduino). Thanks to all for developing such a cracking little app, it's far simpler than *Spice for people who are mathematically challenged like me.

pfalstad, 2021-09-06:
Not sure what you're asking.  But you can go to Circuits->Other Passive Circuits to see a bunch of Lissajous Figures.  Also under Circuits->Op Amps->Chaotic Circuits, a bunch of those have X-Y plots.  Also Misc Devices->Tunnel Diodes->Relaxation Oscillator and some others.

marcdraco, 2021-09-06:
My bad Paul, perhaps mentioning Lissajous gave you the wrong idea, here's an example of the sort of thing I've been using as the central part of a more complex design. 

https://www.jammarcade.net/simple-component-tester-a-k-a-octopus-curve-tracer/

It can be done with a single resistor and an AC supply (below) to generate these tell-tale signatures. It's the sheer simplicity of it that appeals to me - and it's faster than a multimeter at identifying unknown components (both active and passive). I have a bunch of "IN34A" diodes - which are actually Schottky diodes sold on eBay (and other places) as true germanium point-contact devices. This thing identified them immediately with a tell-tale curve (shame the _return by date_ has passed!).

Here's a dead simple one with an LED - it's tracing current vs voltage simultaneously without having to break the circuit to put an ammeter in. The LED is the DUT here, but it's amazing how fast you can identify components (dead _or_ alive) on a PCB without even powering it on.

A practical design would have a range selector for higher voltages (useful for unknown zenners, etc.) but it's really useful to tinker in your simulator and see what sort of values are going to generate a useful figure. 

Using very low voltages (under 3v) and, therefore, currents to "tickle" devices just enough to see if they're OK without risking blowing them up. Paul Carson (Mr Carson's Lab) built one into an old scope and many scopes even come with them built in. It's the sort of thing almost anyone can make and it doesn't even need exact values to work.

Obviously not that sexy in a simulator - I guess you could show several working ones at the same time, but I wanted to keep this to the absolute minimum. 
```
$ 1 0.000005 10.20027730826997 50 5 50 5e-11
v 176 160 368 160 0 1 40 5 0 0 0.5
w 176 160 176 288 0
r 368 160 368 288 0 10000
w 176 288 208 352 0
w 368 288 320 352 0
162 208 352 320 352 2 default-led 1 0 0 0.01
x 161 403 374 406 6 24 Device\sunder\stest
o 5 64 0 4290 5 0.1 0 2 2 0
```

gosh-darn, 2021-09-22:
The schematic in the article you linked to works really well in the simulator :-)
`$ 1 0.000005 10.20027730826997 31 5 50 5e-11
v 640 576 640 496 0 1 50 8.909545442950499 0 0 0.5
r 640 496 752 496 0 560
w 640 576 752 576 0
r 752 576 752 496 0 100
w 752 576 752 608 0
w 752 496 816 496 0
r 816 496 816 544 0 1000
w 816 544 848 544 0
g 848 544 848 560 0 0
w 816 544 816 608 0
w 752 576 912 576 0
w 816 496 912 496 0
g 912 528 912 544 0 0
g 912 608 912 624 0 0
p 912 528 912 496 1 0 0
p 912 576 912 608 1 0 0
c 752 608 816 608 0 0.00001 -0.3816095697099154 0.001
o 15 1 0 4290 5 0.1 0 2 14 0
`

marcdraco, 2021-09-22:
It's a great sim isn't it! 

Slightly O/T but the sim needs adding to the list of circuit simulators on the relevant Wikipedia page. It is mentioned on the Talk page but it's not on the main list. I only found it because one of the electronic YouTube channels I frequent was using it. Blew me away that Paul and Iain are offering this for free! It's an amazing learning tool and so much easier than Spice.

gosh-darn, 2021-09-22:
@marcdraco Absolutely! I discovered it when watching a video by YouTuber Marco Reps. The simulator has served me really well thus far :-)

marcdraco, 2021-09-26:
Marco is hilarious too - I love how he cycles and then takes his two little dogs foraging for capacitors.

I do hope Paul will include a neater version of this in a future release. My ESP32 version is nearly ready - but it really needs a proper PCB as the noise issues are really messing the display.