sharpie7/circuitjs1 #650: Bug npn transistor keeps conducting

sandorex, 2021-09-13:
The NPN transistor for some reason conducts even though it shouldn't, I don't know the validity of the circuit but it probably shouldn't do that?

My guess is the whole issue is caused by the second mosfet and the zener

### Steps to reproduce
1. Close the switch
2. Open the switch

[image]
(code below)

<details>

```
$ 1 0.000005 2.803162489452614 50 5 43 5e-11
R 304 112 304 64 0 0 40 35 0 0 0.5
r 304 208 304 272 0 10
g 304 272 304 288 0 0
f 256 128 304 128 33 1 2.5
f 256 192 304 192 33 1 2.5
w 304 176 304 160 0
w 304 160 304 144 0
34 zener-10 1 1.7143528192810002e-7 0 2.0000000000000084 10 1
z 240 128 240 80 2 zener-10
w 256 80 304 112 0
w 240 80 256 80 0
w 240 128 256 128 0
w 256 160 256 192 0
r 240 192 240 128 0 5000
r 256 224 256 192 0 5000
z 256 160 304 160 2 zener-10
t 224 320 256 320 0 1 -0.13732502925593737 0.6299489553340308 100 default
w 240 240 240 192 0
g 256 336 256 352 0 0
r 144 320 80 320 0 100000
r 240 240 256 304 0 1000
R 80 320 48 320 0 0 40 5 0 0 0.5
s 144 320 224 320 0 1 false

```
</details>

Hope this helps to make the simulator at least a little better

pfalstad, 2021-09-13:
Yes that looks like a bug, caused by roundoff error if I had to guess.

jjdnd, 2021-09-25:
i have the same problem, after hours of scratching my head trying to work out what i did wrong i narrowed it down to this as the simplest circuit that produces the error

set the simulation speed as low as possible and press the button.
the transistor conducts 5.5A for about 2ms then goes back to it should.
id does the same thing again on release of the button.
it only seems to do it if the mosfet DS voltage exceeds about 50v

[image]

$ 1 0.000005 0.00855774612778704 47 5 43 5e-11
f -448 -576 -400 -576 32 1.5 0.02
g -400 -480 -400 -464 0 0
t -160 -592 -112 -592 0 1 -99.70226988881826 0.29773011118173504 100 default
w -400 -672 -112 -672 1
w -112 -672 -112 -608 0
w -112 -576 -112 -480 0
w -112 -480 -400 -480 0
w -400 -592 -400 -672 0
r -400 -560 -400 -480 0 1000
s -512 -576 -448 -576 0 1 true
w -400 -672 -512 -672 0
w -512 -672 -512 -576 0
R -400 -672 -400 -704 0 0 40 100 0 0 0.5
o 3 1 3 x83013 9.94567563433957 0.0001 0 1 0.00009999999999999999 0

kdsch, 2022-09-21:
I believe I have [reproduced](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAsB0AM8MUgzAUwLQHYQFZqRwE4AOMHeZZMSSAJkl0N11lwzDACgAXKY1ygDZcNEEJCt0EWtFpFMheDmSDMawhFizksQYUyCcysJiKlaIeqwAmqAGYBDAK4AbbjxENkxBpH5ifCXAQdBk5fUVYZVV1TVlIWFpYQgTaYnIcMEJVcEEbe2c3TgB3PgEVMrFBYgkSzzEcCxxRbwZYOr9WBK7-SGQLdtLOqH6QPJq+gc4AcxGLSbmoPyD2gCdF8QWF1jAETnWFqw2Knb2h0c3-cUHFrd7Rm8OEsdgao7WXt+fxsSSg3cQnAAxp8qjUfttwBg3rIcMRDEkFIlBHRsFp4FwAEqLI6HV4rKCsHAErQ4faglTg-H9VgCRR1ZpeP4-Gm1XiYCqUkCYQJcnYhGBEXb6QTVPJJSBo-CvRS0ShgZARRrpXKsADOAAcAJZAjC2RyudylHleQImxY3c0LCEPOrmo7m5CNWqlH4OwLvTgAe3AyGCOzkzUssPhjVgSNoKNoUoQmksvrE3ss-IDtA0zHRGOCFj+yCTJmCDDAtDTcQSxFl8roVGImDAwkzsGk4E4QA) the behavior in a circuit that lacks a switch.

```
$ 1 4.0000000000000003e-7 5.459815003314424 59 5 50 5e-11
t 480 336 544 336 0 -1 2.259790053677791 0.23069765531759812 240 default
t 544 384 480 384 0 1 -2.259790053677791 0.24020940285051967 160 default
w 480 336 480 368 0
w 544 352 544 384 0
w 480 400 480 432 0
w 480 432 608 432 0
g 432 432 432 448 0 0
r 432 336 432 432 0 1000
r 432 240 432 336 0 1000
w 432 336 480 336 0
w 432 432 480 432 0
w 432 240 608 240 0
r 608 240 608 320 0 100000
c 608 368 608 432 0 1e-8 2.258652079026427 0.001
R 432 240 432 208 0 0 40 5 0 0 0.5
r 608 368 608 320 0 390
w 544 320 608 320 0
t 736 368 784 368 0 1 -4.591097668602047 0.40890233139795285 160 spice-default
w 784 384 784 432 0
w 784 432 608 432 0
w 784 240 784 352 0
w 608 240 784 240 0
o 13 1 0 12554 2.258652079026427 0.0001 0 2 13 3
o 20 1 0 12291 5 0.0001 1 2 20 3
o 17 1 4 12291 0.40890233142318716 0.0001 2 1
```

I don't know if anyone ever suspected the switch was part of the problem, but this case rules it out.

PhilCS, 2023-01-13:
Optocouplers are affected as well

PhilCS, 2023-01-13:
Potential clue; when editing the NPN's model, setting Reverse Beta to infinity reduces the leakage bug 100-fold (without completely eliminating it) in forward active mode, but amplifies it in reverse active mode.