sharpie7/circuitjs1 #651: Circuit Lagging

renanbertolazo, 2021-09-14:
Hi, is there any way to make large circuits make less lag? I'm doing some large circuits but is very laggy and i can't even use them.

pfalstad, 2021-09-15:
There are some tricks you can use.  Can you send one of the circuits?

renanbertolazo, 2021-09-16:
[Prova8.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/7173940/Prova8.circuitjs.txt)
this one 
and
this one
[Ram_16x4.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/7173941/Ram_16x4.circuitjs.txt)

pfalstad, 2021-09-16:
Those are some big circuits.  I don't have any tricks that would help, I don't think.  I had some ideas to optimize large circuits like this, but I'm not sure when I'll get to it.

renanbertolazo, 2021-09-16:
Ok sr, thank you very much :)

Eugene2003, 2023-05-25:
I am having similar performance issues with my circuit.

li-zhenhai, 2024-01-26:
I used the file "Prova8. circuitjs. txt" from the questioner of this issue for testing，For large circuits, use the default step size of 5 μ s. The calculation is slow, even if the step size is adjusted to 500 μ s. Still slow, use sparse matrix and adjust step size to 500 μ s. The performance improvement for large circuits is very obvious, and I may have forgotten to tell you that large circuits need to adjust the step size to 500 μ s. Even larger step sizes can be adjusted according to the size of the circuit,

A set of test results
Refresh Time (Normal LU Decomposition) Refresh Time (Sparse Matrix LU Decomposition) Matrix Size

2167 milliseconds 414.0 milliseconds 1232
7324 milliseconds 532 milliseconds 1592
32102 milliseconds 618 milliseconds 2352
No test 1441 milliseconds 3511

Using a better sparse matrix structure can be faster and support larger matrices. Below are the test results:
Refresh Time (Sparse Matrix LU Decomposition) Matrix Size

400ms 24470/22100
430ms 48870/44200
460ms 73270/66300
550ms 122070/110500