sharpie7/circuitjs1 #652: LED's don't seem to work as expected

cinap666, 2021-09-15:
I think this must be due to a recent change as I did a similar simulation a few weeks ago. 

The LED(s) have a forward voltage of 3.2V  with 2 in series at 6.4V the max current should be under 20mA the simulation shows 50+mA .     Am i setting up the simulation wrong?  I verified this with real world measurements and get about 16mA.

Example 
https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgpABZsKBTAWjDACgAncMFEPK7Hhp8qVVpDYBnEIOH9aNPCKgqAZgEMANpIZsAbgqXYEvGouWjaVQtF6WYCNtmGqA7gBMOAewAOAHQBHbFtwEGJobBRmVihYGl4ECIRCQgw8E2I8FLAMYhV2MEJeSLEMQhAUZPAyipA3T19A4N4Ie1gUJxcPb38g2yYWsIiolla4hKSUtIyswhy8sTZC3hQaUvLC9dr6nqb+wbbIDq5jFprZavKxDtcZIWULnPKb8EItzZkTcDZbj9PDT52H7cXjyVjFL7sX48QGvARfCS3EqXWFPCrA1ZbSobGodLwqQjCKhrUjIWKtWoCJxKSwQMCQCAAJQYkgAlpIAC7qAB2AGNdEA

pfalstad, 2021-09-15:
What kind of real world LEDs are you using?

cinap666, 2021-09-15:
Everlight White LED  20mA 3.2V forward LK2C-B2832B2B5B2/2T(GC)

pfalstad, 2021-09-15:
This should work better..  It fits the datasheet curve pretty good, though a bit lower than your real world measurements:

https://tinyurl.com/yzgspmkh

I created an advanced model with a higher saturation current, and that seemed to help.

closed 2021-12-10 by pfalstad