sharpie7/circuitjs1 #656: [feature requests] - multiple

pscorbett, 2021-09-19:
Falstad is awesome, thanks for all your work on it! I'd love to see a few things added:

1. Multi-gang potentiometers (or groups like switches have so that multiple pots can be controlled with one value slider. )2. Vactrols2. Pentode tubes
3. Vactrols
4. Bode Plots (I know there is a frequency response in the scope, and you can feed noise into the input, but it gives much less detail).

Thank you!!

carpetbugs, 2021-10-18:
There is https://www.falstad.com/afilter/ for bode plots/frequency response plots, although the number of components is more limited.

pfalstad, 2022-01-15:
I'm probably not going to do multi-gang pots, but you can now adjust the resistance of multiple resistors with a single slider.  Example: https://tinyurl.com/y72h66n9