sharpie7/circuitjs1 #660: I want BUS!

M11a23x58, 2021-09-26:
If I have 32-bit output the fan-out is 32! Too much. Can you make them as one such like output[31:0] and result is 00000000000000000000000000000000 ?

Noitarud, 2023-10-09:
Like this, i  would like to lay parallel wires as a block so i dont have to place/paste 700-odd wires. Specify strands and gap (grid steps) between.

As for the bus request, dropping a wire onto the block of wires could bring up a request which to connect to and the simulator can make them meet, even if the block of wires was narrow that its constituents were off grid. I would like to draw wires and place components in the foreground too (i have needed to place a pair of 12-lane jumpers (moving togetheer) from one place to the next along the bus and i used foreground wires to receive the signals - if i were asked each time which conductor it would get annoying placing the receiving wires or the jumper.

Noitarud, 2023-10-10:
I had a post that said to use macros to convert to code for Circuit. Where is it now?

closed 2026-05-04 by pfalstad