sharpie7/circuitjs1 #673: Add command search menu with "/" shortcut

maximedrouhin, 2021-11-08:
Hi! I would love to see a quick search functionnality. It would show in the File menu like this:
[image]

And it would display a text field, such that, when you type in it, up to 5 (for example) actions matching the typed string best are displayed just below are shown. Such a functionnality is available in many editing software nowadays. Here are two examples.

GIMP:
[image]

Google Docs :
[image]

For example, to add static RAM, I would just need to press `/`, type `static RAM` and press enter.

Due to a few similar option names in the menus, it may be a good idea to not only show the option name but also the successive clicks the user could have made to access the option manually. In the previous example, the first option could be `Add static RAM` with subtitle `Draw > Numeric integrated circuit` and the second `Static RAM` with subtitle `Circuits > Sequential logic`.

maximedrouhin, 2021-11-08:
Sorry, I couldn't add the "enhancement" label.

pfalstad, 2021-12-21:
You can search for components now, but not other menu items like circuits.