sharpie7/circuitjs1 #675: Bug- Inductor current resetting unexpectedly in 3-phase driver

SaltlessLemons, 2021-11-15:
Hi,

I'm trying to design the driver stage for a 3ph BLDC, but my inductors are behaving oddly. Their current is resetting as though they've been removed from circuit, even though they haven't.

[This circuit](http://lushprojects.com/circuitjs/circuitjs.html?ctz=CQAgjCAMB0l3AWAnC1b0DYQGZpLAEyEAc2KxYArNkdiJZPfY5QKYC0YYAUADIjsCxYiAKUs7bJQKjxUeQDMAhgBsAzqyaRuKgQQIJZE4ozFZGMOBLCxiCAhmkB2ZAmwYMyDk50CTRv0Zsd3lLSCwbMGw4YidpDAcCaIRib24Adz05LhlBbIRDbUy8rH8S8EIoDIFgrBosupkimpDPFsaqgHN28AxGSRCwJCauhvAnCTNwYdDuBR6wPrHFoJkbSlFteYGsNvK24PBoDYJtACUx+vKLeQQWUKhj33Z-KZfGHNDYcL1oIjdpCZCJBKEgMEhUuwfN13jgQrD3CILNVylcpikqgAnPTCAKCAwBCywHjFfyfcqfbRAA) is the most minimal circuit I could reproduce the glitch with, but it also happens in my full driver schematic with mosfets, body diodes and resistance on all phases.

The circuit consists of 3 inductors in a wye configuration. Phase A is grounded, phase B is connected to a logic input, and phase C is left open circuit (mosfets connected but both switched off). The idea is that phase A-B is charged, then shorted via ground to allow the current to maintain. The issue arises once this has happened, and both A and B are connected to ground.

Initially current flow continues as expected, through A and B via ground. But at this point if the user makes any change to the circuit, the current is reset to 0. Changes such as creating, deleting, moving or editing a component elsewhere in the document all trigger the glitch, **_even if these changes are entirely isolated from the actual circuit under test_**. Pausing and resuming the simulation does not trigger the glitch, but making any of these changes while the simulation is paused does reset the current once the sim is resumed.

**This only seems to happen when the third phase is connected to it's two mosfets. Leaving the third phase completely open circuit lets the other inductors maintain current as they should, no matter what changes are made to the rest of the circuit.**

A couple of things also seem to trigger the glitch immediately upon the switch from charging to short, instead of waiting for a circuit update.
- Connecting only one of the phase C mosfets
- The inductance of the running phases being too different from one another (6.02mH and 6mH seems to wait for a circuit update, but 6.03mH and 6mH doesn't).
- Putting a resistance in series with the running phases (but only if the voltage across the resistance is above a certain level when the switch occurs. In the example circuit, if the switch occurs while the voltage across the resistance is less than ~10mV, it returns to previous behaviour and waits for a circuit update)

I'm running the offline electron version 2.3.5js, but the glitch also occurs in both current web versions of the simulator that I've tested.
I'm on Windows 10 version 20H2, OS build 19042.1348

pfalstad, 2021-11-15:
The simulator doesn't think one of the inductors has a current path, so it zeroes out the current.  The other inductors don't count as a current path because their currents are not equal (or not close enough to equal).  This is done because if the inductor doesn't have a current path, it causes weird oscillation behavior.  There is probably a better way to fix this.

You can add a 10M resistor to work around this: https://tinyurl.com/ygnuj4kq

SaltlessLemons, 2021-11-15:
Okay no worries, I'll add that into my circuit. Thanks very much :)