sharpie7/circuitjs1 #676: missleading save as dialog

jjdnd, 2021-11-18:
the file "File / Save As" dialog is very misleading

if you press File / SaveAs / OK it does nothing at all

after losing your work a few times you realise you have to click the little blue link to get it to save anything

the "OK" button should read "Cancel" since it does nothing but close the dialog

and the little blue link should be at least a big link that reads "Save" if not a link that looks like a button that says "Save"

then it would act like any other "Save As" dialog in any other program

keep up the excellent work though, it is appreciated, most simulation software is much more difficult to use than this

pfalstad, 2021-12-10:
fixed, now it automatically downloads when you click OK.

closed 2021-12-10 by pfalstad

jjdnd, 2021-12-10:
spot on