sharpie7/circuitjs1 #679: Electrical resolver

jcmk, 2021-11-22:
I teach electronics maintenance in the military. Students often struggle with understanding how resolvers and resolver chains function, and I struggle with finding better ways to explain them. I use the circuit sim to demonstrate many concepts, and it would be very helpful if you could include resolvers.
[Resolvers - Wikipedia](https://en.wikipedia.org/wiki/Resolver_(electrical))
Ideally, I would like to be able to simulate resolvers with up to 2x rotor coil, 2x sine, 2x cos, connected in a resolver chain, a slider for rotating or selecting rotation speed, and a device to convert the resolver output to either analog or digital output.
There seems to be dwindling information on resolvers on the internet, most of what I find are scanned copies of coffee stained textbooks from the 80s, so while my programming skills are not up to the task of forking and implementing, I can certainly provide information on how a resolver is supposed to function.

Thank you.

ipatrol, 2021-12-11:
Maybe this would be best implemented with a generic "electrical machine" parent class, based on the DC motor at present?

jcmk, 2021-12-11:
> Maybe this would be best implemented with a generic "electrical machine" parent class, based on the DC motor at present?

I had thought basing off the transformer class might be the way to go, but you would know better. If you were to base it off a motor, it would have to be AC. I like the idea of an electrical machine class though, as that would pave the way to synchro-servos and the such.

TimFFischer, 2021-12-15:
I tried to implement the formulas given in the [simscape model](https://de.mathworks.com/help/physmod/sps/ref/resolver.html) with dependent sources. This model seems to have the main ingredients for a resolver, but no coupling capacitors an so on.

[Here](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCCsCmC0CcUB0AGATAZjQdjW+AHNigQWJCBgCxRQpRxhgBQAbpSjWlfRpyLCoEQ9CGBQSJtUUkjFs8AGxZFCoUsUjZzAO4cuPKFQP0Uu-QKFGag4WbRgakY5eGR4aVyJCe0AKgBDZgBzEEVDWDBNcNNvezBPY2FIzSpwgSjvbBAAvwBjAB0ABwATAJKAFz8AIz8eFGKSvwAKPIB7AGdm6ABKPxKAM0rijoBLADtuvpLgyr7682caTKWfQ3sUbKTwVMzMmgAZAH0i5jRNkG2rKn3hGgAlE+Z7jI8BAnpYPBsPuMvTVCSeDAkGgsGIWIobQvSJvSIJATfDII2LrQEScGYkF-KGQM4XWGeWC-FKfX6HI4dfHZQnIolRT5MRIgY6nc40sBwqykrwPJ4wzlE7lI2x-QwUSHadkgD6fJQgbCKZLyvlU6WyjJoYSK5IJO4synmAjyyJamXyrAiM4IjU8siaHn0TS5QpFZplSo1Rr5TpTfpDCrzCQhEDA+mKehhjIuUzWmjGh2C0MuWn0XyBI0mpFMz5Isx6BOUTw5otWgVwzJfbgZTSo+gSuJSglJlI1xmpECPNUXbatm6JjujJ7qqQ8nVty4gIcAD2pMqk9HH9BoQ7ZPasjho-fALhXR1n0tp3IZl31RQAFqNnq9EskkVY6wIoUQlP3JO-0DjoTedojq5lHwbAE8UPJNKxPfYQAAOWvWkSSRX5HwgSUQObN4SRPckQAAaznKNfnEassKHABPTNND4ZYEl4fh83NVJhELRC43o+dk3jWJ0yCUIoyZUM3j1OJrU8DU+PtcAUTCHJ8mKd1yiqapvTGSZen9OY6mDAt5U3VjKKtEdRBcccdIpbtslExUFSVcBLNVYTvD2TQAJ8aTqhDFVhHlRD9IubSnLNSDxj8C8r2lZdkkta4QBCudRAdS1IJg6VtjAWy9lsqcjjI5KrHJTIsOwvwOgARwAJwqZpWT8Y4Oh6OcYi8VQhX1VdYrpARLXhZlWTa4lPk6rCarwxMEUErrJ0Kkrysqk5qspOqeIdX4TQffTJPG8DJIwEApoq3Jqj6PJeu5TqotwnLhD47crpAIKYulKNaUImwkz3A8LlEhFjIRPdso+0dLXHWBItu4LLznW10iB9IaEmsqKqqmq6oMnwSCs4Rzn1OHpsR+aIakbhF2swnJzu8GUZwTxx0pydWq0hiJPOU8rXpv94xNJMzCKRnc3-ajS2Qq1uZLDd+b0wWzGqSgMAoYHIDEFBFAoahtrMacoBJDB0lUGwtYgAxO2gDo2gAG1YaBSuYKWvmNaMiSIcBaOYdWvlwDJnGRXV5cnasJiKABXCoCg6VgAlK0YAmqE2jedgRFHpSAnI9yJIG2g2SmgIpoHGDPxiDkOw4jqOY+t7BtsiKhfBQRBxBoNXKBTDBMiVolzgoA3WBOYOCmK4PHiKYO-GD1du4KcZu46Cep8nyeB46VlB+DkpVwAemGMfp83meRnhmaijm2q-EK5oCg37fz63y+Z+U5ooLB0ZphnNeg7Pq+34vjp2i6O+QsfrLn56HVdW8ddZgEuigdwiJFD6zWCATu05u69w6P3ReYx9yj3Hh-S+c8F4dCHh0Ze+5n6n0we-buRRdp7wPn0QqX9b73z-kUZ+zA2jeDsP8YEjstCSHEAQSAit4DGCEcYbAqdvCeF4CwsIjFvA3HEFwqEYgXKKAhMwDAwg+AgAAGKC2BkgRw5BEAYH0duSI0VwbqMoBAHRDlaBmPGP7AAttUC2wc2gDGDkUU20BPEBFGKVKkliMDbRsbECAZj2jQAGAMUYeRRjZ3zu44O7R-ZFBNhMYIwdcJAA) is the full model, or [here](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCCsCmC0CcUB0AGATAZjQdjW+AHNigQWJCBgCxRQpRxhgBQSIAStAM4D2ANgDdoAJxD0KBcBhABrEGgjSIAGQD6AB1oU0INV3A7slEADlwiHRHUAdTZfkh1scA53qAFgEt59CBHYaIABsIDQ6AfpMwcYAHrbgFPTSMc5gSsYAnvFgRtLSGakhNNIAwiUAagDKAKJ8ALbWXGiNGI1UjSjWwuUVNfVtjZCNQY3YjQSN8I1gnVytXGBdPVW1DVxTC80Li1xjXBNc7QtHaV1lK-0LQ8eTHWe9q9N7YAczd90Pl2+b0zt7BydXiMPhc1mANmhZgdOiC+mtrmAtgpRh0xNZGs0MQAqACGdxRjRxWIAxuj0eoACY4ikAFyxACMsVQUJ0yZSsQAKYk8Lgc6AASixFIAZrSybYuJ4AHZ8wUUgDmtMFzNmrK4I0JJPF6g5VNpDO1FJJPNlQtFNOVLPxTXRTVx1sxdrxXDVGq4RNJbN11Lp9MNWMlMoFZqVTKtLttjqJfojLVtXAAjsIaRzo4LicwqDNKCgaLAsElISBYGhICF-Nx+EJRLMkbi0btNZ7KT6GWHWeyuSbgyKxRLpaaFaGVQ2Ru6tbZvfq-bYjdzeT3zZb6LXsTiG3X17Mxx7yVPfeSjYHB0v2xvNfSG-NE8nUwz08wYsWgnmMEEdPB6PnIIgwjRpeoACuNKNAIOLCJ4OL0nw3DMJesAECE+ZUIgTBITM0goMwkJGGA6FoJIVCSLAiKEbIzDsOA+HEXhxakWIDHMi4K5iEgkDYSguHofAOhEXRPGhKYFFUfxOgkdxOgsUxiQMSgbHCbR4lfopeEsfQTFgEgaQoPAQTwOC8CGUZumyax7E4SJSmhDRQTqY4XgcVxxalkUxEuYJEQKUh7nic5ZamdJplyexlGKQWdFIeFUniEF8kWWFRZ8fmiUcBojmWQRhFuaQglqOo6WKXgNBJUVuWqFwXnOTlvkljl0WsSyLLGc1LWIGp8mhd5aB5oV3UBSuqCNa1w1tbJ8kCPIfUlv5pW1ZIvjgI1LK0O17EAO6TXm7kYLmfkhFhG07VtOWzZlYjMJoKGRYlJlzcx52XbdpVXaJ91YRZL0ICEkAEcWJk0Ho6WffRP00b9NCeRZR0RfQOBiapZX5VDu0kQJcOvRDaVQ0ExEvpQOPFlggkeJ4QPLbAeMjslgkmEDRMU8V9NEzQMiPpQb6E5AFBUDtnO8fIEOVoIIjMDw4BGHj6k6YgiSDZxZClgxli5KLPjBDQUuEItrF+K4EBYRgkhGAAYvrhNaVmIQYBb5CICR9mk0AA) as a subcircuit. The behavior looks reasonable to me: When phi is changed, the coupling to the secondary x- and y-coils also changes in a reasonable manner. 
I haven't checked the simulation deeper - a verification e.g. with the matlab model might be beneficial. 

@pfalstad: I'm a kind of confused, that I have to use the factor "400" in order to get the right value out of the derivative within the 6-input and 7-input VCVS. Using "timestep" seems more obvious to me, but does not work... I copied this from the simulation of an inductor. Is there a reason so this value?

pfalstad, 2021-12-15:
You shouldn't...  I just tried creating a VCVS that outputs the dadt of i_p (which is a 100khz sine wave of amplitude 199uV) and I got a 125V sine wave as expected (199uV * 100k * 2pi), so I think the derivatives are being calculated correctly.

The 400 in the variable inductor is just a scaling factor to make the inductance larger (since the input current b is pretty small).

TimFFischer, 2021-12-15:
Shame on me... of course, it's not necessary 🤦
(Un)luckily the non-differential terms were so small, that the output looked right.
The correct output is for small resistor values therefore not far away from the upper solution.

So, I corrected the model: [Here](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCCsCmC0CcUB0AGATAZjQdjW+AHNigQWJCBgCxRQpRxhgBQAbpSjWlfRpyLCoEQ9CGBQSJtUUkjFs8AGxZFCoUsUjZzAO4cuPKFQP0Uu-QKFGag4WbRgakY5eGR4aVyJCe0AKgBDZgBzEEVDWDBNcNNvezBPY2FIzSpwgSjvbBAAvwBjAB0ABwATAJKAFz8AI2KSvwAKPIB7AGcG6ABKPxKAM0ri1oBLADsO7pLgys7zZxpMuZ9DexRspPBUzMyaABkAfSLmNFWQdasqbeEaACUD5muMjwECelg8Gxe409NUSXh-gGAoGIWIobQPSJPSIJATvDIw2LLX4SYGogFfMGQI4nSGeWCfFKvT67PatbHZXHwvFRV5MRIgfaHY4UsBQqyErw3O4Q1l49lw2xfQwUUHaZkgF6vJQgbCKZLSrlk8WSjJoYSy5IJK4M0nmAjSyJqiXSrAiI4wlUcsiaDn0TS5QpFBplSo1Or5NrjHr9CozUL-amKegBjIuUzmmj6m28kDwFyU+i+QJ6g1wumvOFmPRRyiedO5s08qGZN7cDKaRH0EVxMU4mMpcu01IgW5Kk7rBsXaPNoZ3ZVSDkaxunEC9gAe5IlUnoQ-oNF7TPbVkcNC74Bc872E-FlPZNNO2qKAAshvdHolknCrJWBGCiEou5In+gMeDzxtYWXMjfqz8sTuYxLfdthAAA5M9KQJOFPhvCBRX-OsngJfdiRAABrScQ0+cQy1Q3sAE8U00Ph5gSXh+CzY1UmEHMYIjKip1jFwYJ8Hxk39J46VjTijXDBxPBVLjrXABEwhyfJimdcoqlqUo-GGMYum9aYiPXSMTQoydBJcIcVxHfY22yQTZRlOVwBMxVzU8URNC2CtWNyaoQljaiXMYviTmlLYjRAkY-GPU9xTnZJTXOEAAsnGyBFNEDwPFdYwAsrYLNHPZCPiqxiUyVC0PkgBHAAnCoGkZPwDJmcUYi8VQ+W1BdIqpaLqRhElF2yIlXlNfE5x1Qy3OhPMjQGkdctaQritK8rnINT4DWvM1+O8YagNEjAQDGoqGkc7o8gawVYC6sKMIy4QuLXM6QD8iLxRDSkcJsGNN23E5BJhXSWtS9KXoHU0hwO+krpPLSB3SP70hoUbxpKg4ytJCrvsTEhTOEY5tUhzbJrh4HE0MIduB6wHAoRnxcGRkn6Xq7MFVOsiDzNKmbWgg0YzMIoROOT9SI5mK4mYNn82XWmSPAXnqkoDAKAOyAxBQRQKGoNazDHKACQwdJVBsNWIAMFtoFaZoABtWGgArmDFt59VDPEiHATTlbeUnImceFNWlkcy1GIoAFcKgKVpWACAqhgCaoDb15h7cUalIFs52nbWnWSmgIpoBGJORl9-3A+D0Pw-N7A1siKhfBQRBxBoJXKHjDBMjlvFjgoHXWAOP2Cjyv3biKP2-D9hdW4KEZW9aIeR+H4eu9aRlu79koFwAegGAfR+XsfBihzHWm6XKGgKJfV-3lfD7HhSGlA-yTwmccF99vej7vg-Whadoz4Cy+0uvzoZmVqPNbAU6UHcLCRQ2slggGbmOVu7dWid2nsMLc-dB4P0PhPKerQe6tFnlua+u9EH31bkUDaE0YblT8LlJ+p9z5DDfkUa+zBmjeDsN8f4tstBMDYewjhxd2qsV4HQsINFvAXHED4H4YhWKKBBMwDAwg+AgAAGJwWikgRw5BEAYGUWuSI4UgbSMoBABR3gICSwgCML2ABbaoJs-bNF6H7IohtoB2ICEMAqZJdEYDWgY2IEAtEtGgL0XoQw8hDFTpnGxfsWheyKAbUYwQ-YYSAA) is the corrected one, and [this](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCCsCmC0CcUB0AGATAZjQdjW+AHNigQWJCBgCxRQpRxhgBQSIAStAM4D2ANgDdoAJxD0KBcBhABrEGgjSIAGQD6AB1oU0INV3A7slEADlwiHRHUAdTZfkh1scA53qAFgEt59CBHYaIABsIDQ6AfpMwcYAHrbgFPTSMc5gSsYAnvFgRtLSGakhNNIAwiUAagDKAKJ8ALbWXGiNGI1UjSjWwuUVNfVtjZCNQY3YjQSN8I1gnVytXGBdPVW1DVxTC80Li1xjXBNc7QtHaV1lK-0LQ8eTHWe9q9N7YAczd90Pl2+b0zt7BydXiMPhc1mANmhZgdOiC+mtrmAtgpRh0xNZGs0MQAqACGdxRjRxWIAxuj0eoACY4ikAFyxACMsVQUJ0yZSsQAKYk8Lgc6AASixFIAZrSybYuJ4AHZ8wUUgDmtMFzNmrK4I0JJPF6g5VNpDO1FJJPNlQtFNOVLPxTXRTVx1sxdrxXDVGq4RNJbN11Lp9MNWMlMoFZqVTKtLttjqJfojLVtXAAjsIaRzo4LicwqDNKCgaLAsElISBYGhICF-Nx+EJRLMkbi0btNZ7KT6DbYjVyTcGRWKJdLTQqlQ2Ru6tbZvfq-e3jbzu+b+Q26zjF9jl7MRx7yRPfeSjYGB-OV6P6Q35onk6mGenmDFi0E8xggjp4PR85BEGEaNL1ABXGmNAQcWETwcXpPhuGYE9YAIEJ8yoRAmFgmZpBQZhISMMAkLQSQqEkWBERw2RmHYcAsLwzDiwIsRqOZFx6HopBIDQlAMKQ+AdFwyj2NCUxiNIridHwtidHo0JxGohimJIiihNfGTMNE+haLAJA0hQeAgngcF4B03SNIksRGOY1jKKCJTyLMniPE8Yz+JLMtQjw0sig4VQuD4mTnNM4svNE2jEgkozpNggtvPzIs-PExSjPQuzQs48KlNc9RbM80hHJ89KaDUFLYs8tAaASvBCt0NyPNg7CLMyyRIsMlkWT0xqmsQaKpLs4qwo62qUFQermv6lrAqYgR5AKnyHI6kt0t8cB6pZWhfEY4hsA0rAghW3CNJCHqmIAd1GvMvIwXNxu25h9uOw70smyqxGYTR4JCotHuqui7oe-SSzGl78O4iB6Nsn79MgbDi307Kytin6qJB8jQZoCJbMu0z6BwQSFJ4nKkZO36dDRgSeICXKWMoII8PvUm8KwKyvEB+bYAplVi2pmgTEB6mGcKjmWaI28MEfZnIAoKhjsFjj5ARytBBEZgeHAIwKaU9TEESVTqMsXJZZ8YIaCVwhZsMvxXH+5gMEkIwADF-uZ1SsxCDBbfIRB8McWmgA) is the corrected subcircuit...

jcmk, 2021-12-15:
Thank you both for the quick action on this. I've only briefly experimented with it over coffee this morning, but the modelling looks correct. Where it gets more complicated though is in resolver chains. The output of one resolver becomes the input to another resolver, and the final output can then be decoded to calculate the difference between to two angles.

I missed the update where you added the controlled voltage sources, which is a phenomenal feature. I thought the resolvers would have to be hard coded in, but it 

For example, you have a steering wheel that isn't connected to the steering arm, but instead have a resolver on the steering wheel and a resolver and actuator on the steering arm. If the steering arm has the same angle as the wheel, the resolver has no output and the actuator is not told to move. If the steering wheel is turned, now the angles do not match, and the actuator will be  driven until the angles are identical. Aside from the device that decodes the resolver output and drives the actuator, nothing is "calculating" that angle differential. It is produced purely electromagnetically.

I'll do a quick sketch to demonstrate that. It's near impossible to find information on resolver chains online.