sharpie7/circuitjs1 #680: current dots may appear to reverse direction of movement at high speed due to aliasing with refresh rate

ericfont, 2021-11-25:
Take a simple circuit such as a current source feeding connected to a short circuit.  Use a slider to control the current's amperage.  Wen start to increase the current dots seem fine.  However if I cause the current to be high enough to exceed the refresh rate, then the dots appear to be moving in the reverse direction.  Even faster and they return to normal or become chaotic.

This is a bit of a problem particularly when the current direction appear to reverse because the user may assume the apparent direction of the dots is the actual behavior of the circuit.  Of course the user may reduce current speed, but may not realize this, and reducing current speed changes speed of everything.

Suggestion: maybe use another method to indicate that current speed has exceeded the screen drawing refresh.  Maybe put a little warning message.  Maybe change the dots into question marks?  Maybe draw the current as if it was at the maximum unalised speed (like just below (dot distance)/(refresh rate)), that way they are at the maximum apparent speed...and maybe include a warning message or '?' for the dots...or maybe use '>' or '<' instead of '.' so that the direction of the current is always knowable.  Or maybe some other method.

nulldg, 2021-12-10:
What if we add motion blur to the dots?

pfalstad, 2021-12-10:
That's what I wanted to do, but the position of the dots is stored simply as a double in hundreds of places, and those would all have to be changed.  Unless I encode the dot speed + position somehow in a double, which would be a big hack.

Instead, if the current is too fast, I draw it as a blur with a random dot position.  That only requires a minor hack to encode the "too fast" state as a "magic" current position value.

closed 2022-02-19 by pfalstad