sharpie7/circuitjs1 #681: "Add Sliders" duty cycle default max should be 100 (not 1000) and phase offset default max should be 360 (not 1000)

ericfont, 2021-11-25:
When create a voltage or current source and use a waveform like square wave, and then right click it and click "Sliders" and check the boxes for "Phase offset (degrees)" and "Duty Cycle", the form populates both of those max values at 1000 by default:

[image]

But these are not useful or reasonable default max values.  More useful and reasonable default max values would be 100 for duty cycle (cause duty cycle can only run from 0% to 100%) and 360 for Phase offset (cause degrees go from 0 to 360 around a circle).

pfalstad, 2021-11-25:
this is #312

ericfont, 2021-11-25:
oh yes, sorry for the duplicate issue.

pfalstad, 2021-12-10:
fixed

closed 2021-12-10 by pfalstad