sharpie7/circuitjs1 #685: Laplace modules

Fvi589, 2021-12-01:
Hi,
Would it be possible to integrate Laplace modules (1st order, second order, summing, integral, derivative, etc.) in a new library in future versions?
Many thanks for this quality opensource software.

pfalstad, 2021-12-10:
what did you have in mind for the 1st order, 2nd order laplace modules?  You can do summing, integral, derivative with a VCCS.

Fvi589, 2021-12-12:
Hi Paul,

I prepare a small document to explain and I send to you.

Best regard,

Cordialement, 

	* 

http://lyc58-renardfollereau.ac-dijon.fr/index.php/fr [5]/ [5] 

	* 

https://www.cfabourgogne.com/ 

	* 

http://www.plateformeiasp.fr/

  

Links:
------
[1]
https://github.com/sharpie7/circuitjs1/issues/685#issuecomment-991310185
[2]
https://github.com/
[3]
[4]
[5] http://lyc58-renardfollereau.ac-dijon.fr/index.php/fr/