sharpie7/circuitjs1 #686: Impedance

KevinashoUWU, 2021-12-02:
Please add impedance with A + jB form, or polar instead

pfalstad, 2021-12-03:
you want to measure the impedance of a circuit?  or you want to add a complex impedance component to a circuit?

KevinashoUWU, 2021-12-03:
Add a complex impedance

Tremeschin, 2022-01-12:
I'm interested in this feature, it's monotonous and slow having to convert manually reactance to equivalent capacitance or inductance and micro managing two distinct components to get a wanted Z, that would be a big quality of life improvement :)

TimFFischer, 2022-05-22:
I'm also interested in such a feature set. Until now, I tried to emulate an [RLC component with constant impedance amplitude and varying phase](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgzCCsCmC0DsIAMA6ALARgExjJLAnAfFgGwEAcWGEhUUSUcGGAUAO4jxJoiwakKIKr36DkrAB4gaPPlkgEZYAhFhZMIXthABJALYAHaABMAhgDsAxtAA6AZ1u2LAQSMAbAJYAXAK4m7ewBtAHkAC30AXSkZSAxIeUhSWPhEdTAhbSw9I1NLGwcnCwAFMLN7QKCABuiAJRk0JEZ07JoKURxsxkZGmSbu5GQUSFZ6+PhkltjSZs7BnsZSLBRKSH71-oxeAdQRrCREccm5+PF1NEyQQzDPUYaKISmtknkwLsHexgx1+eGYrbAaQUhzQxFeSiyIAAYgAnaAAR180GsAE9Cs4ggAJABe0X2iFIvRoyUJQmwPRyAG8AFoAX1Y+JAhK+clJwkutWgNPpjOZXGybKSWhAABkGQcuARshouO1hMKxYz4FK+N9ePA5fx9sLOdzWGhtUtRAIhNxjeIvlgKBwmTKxEJIBNVRabUa+L1HZNPjaze7GL7yRJGhApbMkGSyB1tsoAPoYGPwGOSMzfAQxyAx1Dx-hxzMoU4KPNNROwA0xhzFAD27GgMIchgAFN4AJT6pAh7WwCOEkBYGVfMC5pBJiipxMZjAoUgxnNYGNz4c0GOD1BNBfl+y+JvNhyWEwOTzbtsQYlhr6QTVCAcxwmZkffGDpvPZ+NzycZxeDlcoJpp4dblsAHpDxbBwTE8MwAHMYTMfQbREGRtQDbUkHFRBlWyIUNV4XpeAAYTQplcMWC5exwkACN5Xp9hIsk5V4TkY0MQi+RNIiyXwYUDCY49lBmGRkniOU+2jHAh3vJAAjjK1hyzGdX3nNY7w0ZSlOHUpykrGF9RwEACAlLtLSdQyg10-S0mVRDjMs1CDRDcM+gjVJHNM+yyQlbBnO+RBUJQCizHcKwDGMcxrGgexBggZIlDw+VGAgAAzGF4T0mRexyJiynsLT0t4CADG5NKIAgEU+iKq4bhkDB0uyRjrk0mF6ASCBdTpLh0ogalEMGbIXAsALK0ggBldgfCsMIsAAUXcfQHCwBwwAcDAHDQWwYVqMxPHcabZvsRb1s27aZpWtaNq2naHFINa+oG4bRu8capuO+xIEuhx4BOmEADU8K+oaLvsZb7AIBwKAcJA1p+v6AaBkHAYh+wIe+37-ue7AlqBj7EchlGYfmwGgaB7onHsfYSe+DZ1nBkmIZW2nsepxnEeGJn5rmtZ+hJ16mfp3YeZJxoaaFnmWYZ0n6YpymmlZkmGwAIwAKjMJwSYAPhV2x4Q1xGdxV+wAH5tdXDYMAALm1jBAIbdQFcMTwFcVswdzFtn7AqdxoCsbwGzMAAaJBfbMK2sFt+25ebZ36ddswFasSt7HlwCMAoJBQ8jkno4V+xPAsRPk9Tu3W0wRg4myD00F4IVqrwgKgtyUKbAihGyfhqWmmQan7EF7Ge4R3YO9J9mqZezu++GAeEe7vvR-H5uEclqWB-x+WlbVpwtfB3WDZpn8TfNparZtu2HaV5sl4cd3Pe9v2A6DhsQ+P8Oz+b3dY-jvOU7T8-7Bj7Pc7lpOn9C6ES2A8V4rQLiPGoMkXgSV4QgMaLMN4DQHLnEuNSQipc+iVzkGA3g1xbh+S+mYda4VPD2G8DlL44AaG2EMOlAYsA0rUPqCedKP08Ko12vjRagM5pLQ7gtWWVhYBOytsbZaKtDDh31N8SU2R+BgGSBQF4WpqHENIdnChVChEJxEWIhsEinDSNbIyLBHptQyl4BgsxOgXhYLwSAOBMQ0DGQ0AkCgkA1DUASLwGUeFPAwisL4HwDh7phAcAANxIRBOWHsHC1BFHhdEuc44WAoZYbwDgACyUELA+H8IESsCUHDBTyGFVs0gCAYCBKCRCcj1AYAhGREAAAVTw+hAh4UrL4GEFQHDFJSVWGsMJfYOC+pWdw3goKBD3A4PCvS4QWG8KwIhJCkmDGquROhzDBhMPeNXYU0pRT0GqtkDhXDdGrUHntfhyNOEA1ej3K69gPowjwrjNGoM3qvO+bczR5DKEwgBi84mQiHANhMFYEw3hYDmBhc2B2MtdxM1WmBMwMKkViwXhsWR-oVRdnkVodR6zYoI14ZC6FsL4Utgdt-MwA9rk0rpfPNuqFIJQGMh6YyyDuiYJ0PwOQ+BmiBlgclO4WDFHJElVsaMCx6A7D+H5IaOdoAAHFkS1jMECnqNCEg7OqtVZo6VDVOOSjQs5FVPAxi+vQY5X0Yz6DMNIUSIAoaXK7rop5QNWgk0Wq-P+DYV5ZKkVYRFD9PBFzkTKpRUABWBmqiqiw6rNUwR1UDXhv8c5BsViGwwYaFYRtbPUSV9iBUp3mMSvZ+lfgjAyMKKERq+D6XzHpEVIANIVFYEAA). Maybe this can be of help.

The subcircuits are: 
- [SineGenerator](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgzCDsB0AcCMAmArAUwLSRABmveYY8ALLMcmNgGxWyJUSJbIjERrr4BQi2WkvELCiD0iYqxAAdAA48+I7CHGKhSienlYqA5RKqk9kgGYAnAI5aQOxEOuHhEgGoB9ALYBDAB48k13cK0wsLw4CAeAFQAzgCWAHYAFAkARhEALrIAxgCUEYgRMjHZVjasSgaO6iAyABYxLk5cQA)
- [CalcImpedances](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgzCDsB0AcCMAmArAUwLSRABmveYY8ALLMcmNgGxWyJUSJbIjERrr4BQAHiOogCcgkPFgQBVeKNhVWoxCAAKACwCGAZwD2AJ16jIsEEMaJixwXPNIQAYTUAHNQGMAlgBcAnvrER4Vc0QwOX8sa0UASQA7ABMAV2d3VwA3D29EbCwBWCMwDP487HB88wiAWwBvAC0AXy5-IoFkOUI5bNzgnGMi+Gw+-oGuACV+RBzwZEV0QonFIqLiea756GRh0chFMEmNrbIuxRQBkBYl3DWMrJQWzoEzcE7SsoB9B3VtPUvR5uN7u8D7uYhqhXu9dOt2sZFgV8mYzqx4Ss1g1vnJEH8xkZ0eYivlegMCVwvpCMkZprDsEYgahqnUvtdirjxjNzLYiZlRFQittFPgtjtzAAzHQAR3ZWBRMyQuRKIHKtK4AHNjD8eSqWvt5kSbJKKTKlooABQAIwAVGoADoaAB8FpFVuwAEorQB+B3QAnYeAALit8AA9IaBKaHK5TWa1I7xerRIJDuN4HH5AAZaMo7GiGwZoWitNc2O8nVJp4K5UMjPl6Far7+VhU+vyYEKmtozZ14xtku1bW8-OJw6UgsHEAaVAAG1QiUNagANNgZ2pA4gQ2HjY6o8TYEUt-w+SAd+Y3q5o+gD2S96R5FUTzumFM93erzfudhzJwbJRzIeVMfEDZTwsZJ5O2uIgGoprOFoGgmv6YjYCuUbKrWO61nuWp8Jg0hBEYWHENI4RymUVpKGoOjuFahoaK4USoIhnLgK+9GtMsPb8Lebbkh2czGGBppUVEMFwQhXAAO67LM4lwjg+iYH4LAEDI8g2MCxGkeR0GQfxtFcEAA)
- [VarResistor](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgzCDsB0AcCMAmArAUwLSRABmveYY8ALLMcmNgGxWyJUSJbIjERrr4BQA7iMpCo5+gkFWI5erWNhDpxIofInYppWQvXhEkxNixbZbHUZAAlLogLTZ6MDuTEJdk+BAAKAMboAhgEoAendcbGx4AB0ABwAjPzUZOXsbOQVVPgElBQzEk3iNCWzUy31k5VL8HQko4oMEzQSkWFY5LiA)
- [VarLC](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgzCDsB0AcCMAmArAUwLSRABmveYY8ALLMcmNgGxWyJUSJbIjERrr4BQi2WktEOirEogzsVGiAOgAcefKMkQh4AThWRlIFFVZCuAdxCwRINXtOiz2IydLnLZizjtlYqxB6uePt41oq8FTYSkEargHaZoGOkSZmMdGitogEYXEIei6hEAAUACYAxgUALugFAIZlAJQAVABGCvzaSB5UiKJt+gDCbmbdPuoqqYpUkCMgHSn6ADJcAOZTE+Aq4yoOoanp6zhQvHuhKpX9ouguPsIpdpBgeudOZzk3d0Jgmq8P8QIe6O9Qn3+-gSZ3+PhsPBIYl+Zlg2FCVz2oiqZUap1UEViw1cvH4glCAgRwX0ADk+kA)

The limitations are:
- It does have difficulties when the phase angle is near 0°
- The settling of the output depends on the available resistance (for phase angle is near +-90° it can take "forever")
- It sometimes starts with an sinular matrix error (e.g. when the simulation above is resetted). Resetting twice helps here.
- It's not fast

pfalstad, 2022-05-22:
The problem with doing anything related to impedance or phasors is that I need to know the input frequency, which has to be sinusoidal.  I suppose could have a check to make sure that the circuit only contains one time-varying source and make sure it's a sine wave, so I can get the frequency and do the calculations.  If that check fails, I could generate an error.

TimFFischer, 2022-05-23:
One could also have multiple time-varying sinusidal sources, when the frequency base is fixed. Not sure, whether the "fixed frequency option" is better to be selectable on the sidebar, as an option in the dropdown of the source menu or as an element in the option menu. Maybe, it's even better to have this type of simulation in the "afilter" instead - then the bode diagram and pole zero plot can be used (but this simulation would need additional things like multiple outputs, current measurement, etc.). 

Generally, it would be great to have additional components in this type of simulation: 

- phasor diagram (static or optionaly rotating)
- complex voltage and current source, impedance and complex measurement tools (Amplitude + Phase angle or Re + Im part)

Enough for my "wish list" ;-)