sharpie7/circuitjs1 #687: Bug: circuit disappears when started

TimFFischer, 2021-12-08:
I just played around and stumbled over nice circuit, which disappears when it is started.
Once stopped it appears again. The circuit does not make sense, but neither does the disappearance ;-)

The following link shows (or does not show) the circuit. Once the simulation is stopped the circuit gets visible.
Somhow, all components seem to be needed for the bug to happen and the resistance has to be high.

[http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAWwpABYqEBTAWjDAChtC8Rs1xiUPPmm6QobAE60whEChpiaM8ClHJ4kNgHdps+WJTy5C8TsM05+OUf2nwNC2AyywA8M-EATexdXcwDu564ABy2DRsAOY+cnj+gbgWYpoASjyQjqrGYmBZybQ5eNBU+TAI2jycKtxKLnlsAIbWjm68OW7E4EgsSGAacOBFKKQYNEThGPKGSNjQeBh4pHi5YNgYCH00zCgQfRoVroJOekbH4gBm6RZ4YongsiZg0EgwkChsAC5C7YJt1VAgFiA7CzXBgAikBACQgTYGgyBgBCTKHYYgIBAuAaeBjnBoAVwANh82EA](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAWwpABYqEBTAWjDAChtC8Rs1xiUPPmm6QobAE60whEChpiaM8ClHJ4kNgHdps+WJTy5C8TsM05+OUf2nwNC2AyywA8M-EATexdXcwDu564ABy2DRsAOY+cnj+gbgWYpoASjyQjqrGYmBZybQ5eNBU+TAI2jycKtxKLnlsAIbWjm68OW7E4EgsSGAacOBFKKQYNEThGPKGSNjQeBh4pHi5YNgYCH00zCgQfRoVroJOekbH4gBm6RZ4YongsiZg0EgwkChsAC5C7YJt1VAgFiA7CzXBgAikBACQgTYGgyBgBCTKHYYgIBAuAaeBjnBoAVwANh82EA)

nulldg, 2021-12-10:
Circuits often stops being rendered when errors are occurring on each frame.

Checking the matrix dump `runCircuit` produces for this circuit, there are some infinities and NaNs appearing in the circuit matrix. Normally `runCircuit` would catch this NaNsplosion and halt the simulation, but this debugging feature is skipped due to this circuit's matrix size.

To be specific about the source of the errors: When `lu_factor()` is called, the absurd matrix values passed to it result in the local variable `largestRow` not being changed from its default value -1. See lines 5322 to 5344 of `CirSim.java`: The assumption that  `abs(q)` will always be equal to or larger than zero is broken when `q` is NaN, therefore `largestRow` is not guaranteed to be set under this edge-case. This largest row index of -1 is then used to index the matrix during pivoting and this ultimately results in "undefined has no properties" errors the console is throwing out every frame.

I *think* the circuit elements are working as intended (despite this circuit's unintended behavior, the circuit is frankly absurd), but `lu_factor()` might need to be modified slightly to withstand NaNs appearing in the matrix. Triggering a singular matrix error is better spamming the console with uncaught exceptions which prevent rendering and cause all kinds of unforeseen problems.

@pfalstad For reference, this is a minified version of this circuit (which will trigger the NaN/Infinity check).
```
$ 1 0.000005 10.20027730826997 53 5 43 5e-11
a 288 176 368 176 9 15 -15 1000000 0 0 100000
t 368 176 368 240 1 -1 0 0 100 default
r 288 192 352 240 0 100000
R 384 240 416 240 0 0 40 18.3 0 0 0.5
```

pfalstad, 2021-12-10:
I've fixed this already but haven't uploaded the fix yet.  Yes, nulldg, that's exactly the problem.  Although I didn't change lu_factor to withstand NaN's, I just fixed it to abort if it sees any.

nulldg, 2021-12-10:
Oh, as usual I am hilariously slow. This issue is open and I didn't see any commits so I figured I'd give whatever help I could. Still, I'm glad the issue was resolved.

pfalstad, 2021-12-10:
Thanks.  If an issue is assigned to me, that usually means I am working on it or have fixed it but haven't uploaded the fix.

TimFFischer, 2021-12-13:
Thanks for the fast correction. 

Interestingly, it works, when the circuit was at least running once - but when clicking on the link above (i.e. For the first time of running, it needs a mouseover to Show the circuit. It's a minor bug, but more visible from mobile devices..

closed 2022-01-15 by pfalstad