sharpie7/circuitjs1 #695: Optional circle around transistors

K0rrid0r, 2021-12-15:
It would be nice to see transistors with the circle around them it's just to easier spot them in bigger circuits.

[image]

The circle could be an option in the window
[image]

The triode somehow has this so why not transistors?
[image]

thoughts? objections?

pfalstad, 2021-12-16:
You probably want this to apply to all transistors in a circuit, right?  Rather than have a separate checkbox for every one?

K0rrid0r, 2021-12-16:
yes i think that would be the correct option.

K0rrid0r, 2021-12-21:
I see you completed it, I'm eager to se what it looks like :D

closed 2022-01-15 by pfalstad