sharpie7/circuitjs1 #697: ne555

woreno, 2021-12-17:
if using the 555 IC i cannot do the most simple circuit - monostable (one shot trigger). 
i was expecting a low-hi-low output when flipping the switch;
also when the cap charges to 2/3 vcc i expected to go down (it won't).
however, opening your 555 internal example (2 opamps + 2 nors), and modifying the 
pins to the monostable circuit it works as it should
maybe my bad?

-----------------------------------------------------------
monostable:
6. 7. threshold-discharge: shorted; between this node a resistor to vcc and capacitor to ground
4, 8. reset, vcc to 5V; 1. gnd
5. control don't care
2. trigger: connected to a pull up resistor to vcc; followed to a switch to ground 
won't make a signal low-hi-low by the 1.1RC.. 

this is the example with the IC:

$ 1 0.000005 5.023272298708815 64 7 50 5e-11
g 16 192 16 208 0 0
r 64 192 64 112 0 10000
R 304 32 272 32 0 0 40 5 0 0 0.5
O 352 160 416 160 0 0
165 176 96 192 96 6 4.999999950000001
w 240 64 304 64 0
w 304 64 304 128 0
w 352 160 304 160 0
w 64 192 112 192 0
g 272 256 272 272 0 0
w 304 64 304 32 0
s 64 192 16 192 0 0 false
R 64 112 32 112 0 0 40 5 0 0 0.5
R 144 64 112 64 0 0 40 5 0 0 0.5
r 144 64 144 128 0 1000
w 160 128 176 128 0
c 144 240 144 272 0 0.00001 4.974491983331485 0.001
g 144 272 144 288 0 0
w 160 128 144 128 0
w 176 224 144 224 0
w 144 224 144 240 0
w 144 224 144 128 0
w 176 192 112 192 0
o 3 32 0 x83013 5 0.1 0 1 1.25 0
o 16 32 0 x83013 5 0.1 1 2 1.25 0 16 3 0.025 0
-----------------------------------------------------------

this is the same but using your internals example :
$ 3 0.000005 5.023272298708815 64 7 50 5e-11
a 288 168 384 168 9 5 0 1000000 0 3.333333333333333 100000
a 288 264 384 264 9 5 0 1000000 6.666666666666666 9.703215267284776 100000
r 240 56 240 104 0 5000
r 240 104 240 152 0 5000
w 240 152 240 280 0
r 240 280 240 328 0 5000
w 240 152 288 152 0
w 240 104 272 104 0
w 272 104 272 280 0
w 272 280 288 280 0
w 464 176 464 192 0
w 384 184 384 192 0
w 384 192 464 240 0
w 464 240 464 256 0
w 384 240 384 248 0
w 384 240 464 192 0
R 240 56 240 16 0 0 40 10 0 0 0.5
R 56 56 56 24 0 0 40 10 0 0 0.5
r 56 56 56 120 0 1000
r -16 144 -16 208 0 10000
w 216 120 216 352 0
c 56 120 56 152 0 0.000009999999999999999 9.703215267284776 0.001
g 56 152 56 168 0 0
r 384 368 464 368 0 10000
w 464 176 496 176 0
w 464 368 496 368 0
153 384 256 464 256 1 2 0 5
153 384 176 464 176 1 2 0 5
w 496 176 496 368 0
O 464 256 544 256 0 0
t 328 368 296 368 0 1 -9.703215169252609 9.803216694199141e-8 100 default
w 216 352 296 352 0
g 296 384 296 408 0 0
w 328 368 384 368 0
x 120 115 189 118 4 16 discharge
x 131 173 176 176 4 16 trigger
x 120 242 185 245 4 16 threshold
w 272 104 272 32 0
x 272 22 320 25 4 16 control
b 514 398 200 40 0
w 240 328 240 384 0
w 240 384 296 384 0
w 56 56 -16 56 0
w -16 144 -16 56 0
s -16 208 -72 208 0 0 false
g -72 208 -72 224 0 0
w 56 120 120 120 0
w 120 120 120 248 0
w 288 248 120 248 0
w 216 120 120 120 0
w -16 208 200 208 0
w 288 184 200 184 0
w 200 184 200 200 0
w 200 200 200 208 0
o 21 16 0 4099 10 0.00625 0 2 21 3
o 29 32 0 4106 5 0.00009765625 1 1

woreno, 2021-12-17:
btw, when copy-paste the internal 555 example to a new blank circuit,  i could not connect to the wires
(maybe due to the grid that was different) so i had to reinsert the components and set the values
accordingly

pfalstad, 2021-12-17:
Wikipedia says "As long as this pin [the Trigger pin] continues to be kept at a low voltage, the OUTPUT will remain in the high state."  I have not doublechecked this against the 555 schematic though.  My 555 internals is a simplified version; I'm not sure if the real 555 acts the same way.  I can check against the transistor-level 555 schematic.

The circuit works if the switch is normally open.  If you close it for a short interval, then you will get the monostable behavior.

pfalstad, 2021-12-17:
The copy-paste problem is because the internal 555 example is using the small grid option.  We should turn that option on if pasting something from a small grid circuit.

woreno, 2021-12-17:
thanks, i find odd that with a fast strobe (hi-low-hi) on ~TR, the cap when reaches 2/3vcc (with ~tr=hi) the output doesn't go low.
as said before, your example internals circuit doesn't behave like this; at least for me.

all the best

pfalstad, 2021-12-17:
After looking at a better block diagram, and the full transistor-level schematic, I think it's a bug in the 555 internals circuit.  The output is low but the discharge pin has no current.  That shouldn't happen.

woreno, 2021-12-17:
ok ok ! maybe my bad. i will look for that option then.
great work.

pfalstad, 2021-12-17:
I'll keep this open because I should fix the internals circuit.  Its behavior should match the 555 component.

woreno, 2021-12-17:
both seem ok now.. for sure it was me, just stuborn'ing flipping the switch low-hi-low instead the opposite hi-low-hi,
then swichting between two diagrams. 

one note, the D flipflop you can put reset/set inverted, but then, the IC drawing doesn't reflect if R/S is active to low or to hi
i used to follow the 74 version, so maybe i don't know 
sorry so much for the time, great work.

pfalstad, 2021-12-17:
Re the D flipflop, it shows a little bubble if R/S is active low

woreno, 2021-12-17:
argh ok..forgive me

17/12/2021 à(s) 21:32:

closed 2021-12-21 by pfalstad