sharpie7/circuitjs1 #699: LM317 output incorrect

tekinozbek, 2022-01-04:
I'm getting 1.7V output for a classic LM317 5V setup:

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgpABZsKBTAWjDACgawqxiVxeQKBIUHDwIAH4AZALLYw5SAB0AzvKrK1XKKpS4dKsGgPC4Bo5TOaw2JNeIbVRmldXZCrlQjyfWjr1z+Fp4oLv6EGJ6EPkFongja1pHmHgZx5ri+kHaqNAYwEJosqkUwCGjEHjgYCLbC2HhOcKoAJgwAZgCGAK4ANgAuBsUqxNAKYDQehA142HTCqkxlFVW2BEZ4YI2GzSptXX2DRWCqeNCEhAnYxJwoeHiEKCiLy9eQxGC1NBgoNpVNmn2PQGQxOKjONASeEhxGwGEmd2ICBe0HKHwQxHW0KEkGw2ABrQ6wKOizBEJqkD0CgQGJ4NBR302VMInwxCAwrHxO0BRMOoJKqN+ejmkJo0Nq2yWqK2k3eDXZSOExAJe15IOOArwmMh7lhGE2UzBUomzmyczAhDopGmKqBfI1Khg0RINQwlpoLjIeJRwg8RnelEmHsItrVJJUwxgNVqrBInzWdElMG+HsxvExT1qkAUoYO6tJTjG91s0S1SMgGCRGBRkEt03eJFqFz0j1zxP5jtg0MeF3KeL+IoZNkgNCRn0e2fWbftBcdQ9wo4SwjQW1+0-zEbJ0BovzwFLmmMrzwjZWidAtZB8bv11e5hLz4aNo3GsumxfmIc75V+HoeFqwjw8Ouj4CtmvzXByRDyJwpBkAKtR4jgnybJ8SLbIE97ttYKJwj8aAuB6nyfD8yKdt8mJbPgI5ungCREMB5iktAQgoIekR4iQeKUoWNKVLwHjXN4i4YMwt4YaqD6MRGozZnMsxkO8oSWqRZQ7rMYpKgoYSfuJdobkaUY0vIF7xkJDQCpwSpCA47w2T4pG6WGUlSvJsLqQ0ER6D8Fk2MIFqUJy9yXGJux6SBJ7QJBswcrieK3HQFnRjw+oXBaEQTAx2GdriNjRAoNQRJEkzwbi9xcB8LgWrMOmhU5WUwDl7ibFgwhYCOn5lLWlRpVwlCAWCjmSVlTAQru+6jnuvDwdENwNA4HxQQsd4SVhTiLGcpCUKEOCPEiJACoBlRGYyPBkCFPJDWt0nnBeDiYFwAEPAd1mRNC9wOC4GIMceEgAF4AG4APYDJ0ADmDBKEoACOhDQPQP0A8D-RgxD0Ow-Dqh-UDIPg5DMNwyA2DHkwKljOgORqMwrACtGxlxrUZnbA12YGNgVMDWMNyfJMWBuvglxgszhSqAgVPHlGGxwlweLmqEXJCx2pSome8guleER7sBbAAEr8HwPB8D4etQCbLjgHwGhQKibAAO6CPcxtNcbkBsAAToT0SO57C4m6ELvu07PtO2EJs-C7duByOogiD74fR4TUdCCISdQGwoMe3gtBmMHlomy76cIB6xuXCIBt527GfG4XeRl1QXFsAA8pXZdO5U4jsAAHrQFrgI8tA3L3SB5KEIDawAFL0gOdC0ACUbBAA

Additionally, sometimes the component will output values < 1.25V (for V(in) > 1.25V of course):

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgpABZsKBTAWjDACgawqxiVxeQKBIUHDwIAH4AZALLYw5SAB0AzvKrK1XKKpS4dKsGgPC4Bo5TOaw2JNeIbVRmldXZCrlQjyfWjr1z+Fp4oLv6EGJ6EPkFongja1pHmHgZx5ri+kHaqNAYwEJosqkUwCGjEHhgJ2Ng0nBg0TnCqACYMAGYAhgCuADYALgbFhjCQNNE0jUQI2D5gqkxlFVU0xDTemMQLo5rt3f1DRTvE0Agu5YQ8kHqEE4SLy9gO29FgNHgYeM17nb2DwxO0A+kAiPju5xQeCmjzOKG2CGIeDmGDB+GIPzaf0OgNUp0oDjwCBwXwIkAxKiWwK+RnkhAihGwGCMQkxKn2-yOix2y1pdUhz0wKFhCWhHmIwkIs3pSLZHJxxxK0DwSI2TMg8j0hIelIKnFC2XK53w8OJcuxAMVKjKdUJ9VBiJsNxFd0ZLglvGm63NB0t3KViO8TI8GxIpGysKm5TIQjwERwQp9nNxhlgtihIaZRFYYG+utgdyEd14cfGUzwecCWN9XMpPNgDhQejwXAlREgYGikZsH1YKC4jMq1STCv91u7dRbRgHzwiCBHfrreOBswQCWNzNmGxda8+dSbbau2AXtZ2LDGE2h0ylcw7Aab7w+OHGpBP5iVHf7zwwaHTFmhFI2rUOAKBs2zfhSVbshap6wkyP5oC4nBGNCzrWtSSK5vgcw0BU5THrs1bJtY3INhKuFzDYQbCE4ZwEoQQjYFCIJMQgzAYG+JF1he1QTGubqqoBZzuE2haxi2nxNIR0E1u++aBsiqLjMIpAOPO6GcFcHigkQPjeLwnFOLC0QeFcnBrk2iL9kqnDPMQTFSlCcCfDR0nyouZ5lIirxrmu7zVOUNnVO46wVv2VH0oZhgfkICgEBEpC4FwLYcRpqoqoiXyorhjRSVB7mwehNwJF8RAYIlHb4AoAZwIlvDeB4UIGW5MFyUwpxqn5G5gFu6llPS5ISratgOql+WtVx56wKJHxIjclRrkJZl0gopqNHABHjbJk1cNAjIkFKPWsbONkpYEk4kIg5xvsKKgSAAXgAbgA9oMXQAOYMEoSgAI6ENA9C3Q9L1vZ931-QDgiqMDr0DB9X2-f99BMYsfXQIEQgGNgzCsAGEqKSGKnhmjcCFG4OM7GAaa+RMnZfEiGYfh2Jg47dXmkJ2vkJFGgVFczVqlGcUreDguGmtmm0tDJnJsAASvwfA8HwPgK1AasuOAfAaFAZxsAA7oIFaq+4eCq5AbAAE4gCbxvRNb4xq4E5tWzbuB5DbYSO-A+vW3bbuiCI-vmwbRb23koeh+b72+6bnse3catRxQ9Sq1KIhK4nlsx6r5x5BnVC1ObADy2cZzblTiOwAAetCduADG0Os9dIOHeSywAFH0z1dK0ACUbBAA

pfalstad, 2022-01-04:
hmm.  try this one: https://tinyurl.com/y2m6bltb

tekinozbek, 2022-01-04:
That looks righter :)

I guess just a bug with the built-in module.

pfalstad, 2022-01-05:
ok I fixed this but you will have to re-create the LM317.  the old one won't show up.  I assume this isn't a big deal since obviously no one was using the LM317.

tekinozbek, 2022-01-06:
Thank you, verified. Much appreciated! :)

closed 2022-01-06 by tekinozbek