sharpie7/circuitjs1 #700: Managing subcircuits

Ceven04, 2022-01-04:
Hi
So i created a couple of subcircuits on the offline version
But one of them was wrong and to correct the error i created another one with the same name but this time it works as expected
Is there a way to delete subcircuits of the menu?
Its very easy to confuse them since theyre identical in structure

pfalstad, 2022-01-04:
Unfortunately the only way to do this is:

1. Options->Toggle Dev Tools
2. Click the >>, select Application
3. Expand Local Storage, select file://
4. find the offending subcircuit: key, right click on it, Delete

Ceven04, 2022-01-04:
Thanks that did it
It would be nice to see an option to do it from the subcircuit menu in the future