sharpie7/circuitjs1 #701: Very slow when input to VCVS is zero

carpetbugs, 2022-01-06:
When the input of a VCVS is zero for a very large circuit, the circuit slows down drastically. 
Example: 
https://tinyurl.com/y3zxq8on

pfalstad, 2022-01-06:
Yeah that's pretty strange.  When the switched is flipped, the output is always zero, so I'm not sure why it has to be so slow.

ormaaj, 2022-02-28:
I wonder if it has something to do with the implicit zero volt reference for the dependent sources. This is another reason I was thinking about the spice `v(refdes, refdes)` syntax. It may do some fancy things to deal with float error by effectively level shifting relative voltage measurements so that float values are aliased to the same base value before doing comparisons (that might not actually be possible to deal with in javascript). I didn't see this simulator doing anything fancy with that. Here, `a` implicitly seems to become `a - x` where x is whatever the global 0 volt reference happens to be. And passing a reference voltage into b and writing `a - b` becomes `(a - x) - (b - x)`.

There could be situations where this makes the matrix solver do extra work. I know of a few that result in singular matrix that could be theoretically resolvable. A circuit that completely lacks a ground node could use the implicit ground in an expression to resolve what the zero volt point must be for instance.

This simulator does pretty well. Many struggle with this kind of thing. (I don't even know why this works. I guess you have a global rshunt to deal with the floating ground.)

https://tinyurl.com/y8d9ewy9

Delete the ground node and it places one somewhere that doesn't quite satisfy the constraints. Many variants are singular matrix as well as voltage source loops that could in principle be resolved, but you need to resolve perfect arbitrary precision comparisons to do it.

pfalstad, 2022-02-28:
I don't think so, since if you explicitly ground the B input, it's just as slow.

ormaaj, it's funny that your circuit works better with the ground element floating there.  When that ground is there, the simulator uses a different set of logic to pick the ground node for the subcircuit on the right.  Without it, it picks the low end of one of the voltage elements.  With it, it just picks a node at random, which is a lucky guess in this case.