sharpie7/circuitjs1 #703: Bug - Latch - Small grid export

ArmoredRodent, 2022-01-11:
When a latch positioned on a small grid gets exported, it gets imported on a large grid.

$ 3 0.000005 10.20027730826997 50 5 50 5e-11
164 704 488 712 488 1 4 0 0 0 0 false 0
166 808 488 840 488 1 4
168 712 664 736 664 **2** 4 0 0 0 0
168 816 600 832 600 **2** 8 0 0 0 0 0 0 0 0

The problem appears to be with the number 2 highlighted.  Change the 2 to 1 and the circuit is imported correctly.

$ 3 0.000005 10.20027730826997 50 5 50 5e-11
164 704 488 712 488 1 4 0 0 0 0 false 0
166 808 488 840 488 1 4
168 712 664 736 664 1 4 0 0 0 0
168 816 600 832 600 1 8 0 0 0 0 0 0 0 0

ipatrol, 2022-01-15:
I don't know if it's the same problem, but a similar issue also appears when trying to copy or duplicate latches. They are always pasted in the large-grid form.

pfalstad, 2022-01-15:
it is the same.

ArmoredRodent, 2022-01-15:
For ipatrol: Sometimes even with Get Latch on small scale, the latch is initially small, but if you move it a tiny bit, it reverts to large scale.

closed 2022-02-19 by pfalstad