sharpie7/circuitjs1 #705: can't slide slider all the way down to it's minimum value easily!

ericfont, 2022-01-14:
I have a 0-to-50k resistor with slider named "Tempo" here: https://tinyurl.com/y83zccpo 
([circuit-20220114-0010.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/7868007/circuit-20220114-0010.circuitjs.txt))

Unfortunately when I try to drag the slider to 0 with my mouse, the value only seems to be able to go down to either 500, or 1k, or 1.5k, or even something higher like 7.5k.  I can't get that slider to be slided down to 0 reliably.  I'm even on a big screen (2560x1440).  Though I can reliably always slide the value up to its maximum of 50k.

(I'm on arch linux 5.15.14-1-lts running latest kde wayland running Google Chrome Version 97.0.4692.71 (Official Build) (64-bit))

ericfont, 2022-01-14:
the workaround I discovered is to set the minimum slider value to 1n ohms.

And I think I know why this glitch happens...  The reason is because there is probably some internal condition in circuit js that prohibits resistors from having 0 ohms, cause it seems I am simply not able to set a regular resistor to 0 using the popup set value.  However, circuit js does allow me to set a minimum value of 0 for a slider for a resistor.  And so when I try to move my slider to the minimum value, the value that ends up getting assigned to the resistor is relative to whatever pixel x-position my mouse was at in the frame *prior* to reaching the slider's minimum position...

ericfont, 2022-01-14:
I do consider this a bug and I've renamed the title.  I would say that either the slider edit GUI should simply prohibit me from setting 0 as the minimum value...in which case if user tries to put zero, maybe the GUI should force it to something very small like 1n ohm.  OR circuit js should allow 0 ohm resistors.

ericfont, 2022-01-14:
I notice it is possible in the slider configure window that user can set minimum to negative values (which of course are meaningless in reality).  So if were to do input validation for resistor sliders, maybe do if (min <= 0) then set to 1n.

ericfont, 2022-01-14:
a similar phenomenon occurs with capacitor slider...setting min to 0 means that I can't slide all the way down to zero.  (But while 0 capacitance is practically impossible, I do think should allow 0 capacitance.)  But if your design wants to prohibit 0 capacitance, then maybe similar to resistor suggestion (1) maybe just set minimum to very small like 1p.

ericfont, 2022-01-14:
(I'm noticing setting min for capacitor slider to negative results in capacitor slider simply only allowing user to set slider to max, which I suppose is a safe coding decision.  A similar thing could be done for resistors set to negative values.)

pfalstad, 2022-01-14:
I suppose I could change the slider GUI to set a small minimum value for resistors or capacitors.  Allowing zero valued resistors or capacitors would not make sense, since those would have to be implemented a completely different way than a small resistor/capacitor.  Probably not worth the trouble for such an unusual case.

ericfont, 2022-01-14:
I understand, sounds good, thanks.

pfalstad, 2022-01-15:
Actually what might be easier and also make more sense is for the resistance to be set to a small value if you try to set it to 0 (with a slider or otherwise), rather than just doing nothing.  Similarly for a capacitor.

closed 2022-02-19 by pfalstad