sharpie7/circuitjs1 #706: [Feature Request] Logarithmic Sliders

ericfont, 2022-01-14:
Currently sliders only have linear scale.  I would like an checkbox option in the slider view to also have logarithmic scale.  This can help simulate logarithmic potentiometers, for instance.

ormaaj, 2022-02-28:
Also log voltage visualization perhaps? Would have to see some examples to know if it's useful I guess. It could end up with too much of the color range squished at the high and low ends.

pfalstad, 2022-02-28:
That doesn't sound that useful, but if you really want it, you can file a separate issue.  🙂

ormaaj, 2022-02-28:
Meh don't feel strongly about it so didn't. :D Alternatively, optional extra colors to extend dynamic range. I think there's already an issue for that.

Good idea with the sliders though. Same potential problem with coarseness of the adjustment at the ends. An adjustable log base could tune that maybe. Lots of knobs = feature creep. I have too many ideas.