sharpie7/circuitjs1 #712: NPN transistor is closed, but throw current

MikePryadko, 2022-01-21:
[image]

`Vbe` = 89.7 mV, `Ib` = -17.94 uA, but `Ic` = 9.185 mA -- it's impossible :neutral_face: 

<details>
<summary>schematic text</summary>

```
$ 1 0.0000049999999999999996 10.20027730826997 50 20 50 5e-11
f 784 432 848 432 32 1.5 100
r 848 448 848 512 0 8
g 848 512 848 528 0 0
R 848 416 848 336 0 0 40 29 0 0 0.5
r 896 448 896 400 0 1
c 896 400 896 368 0 0.000001 -9.276152783846982 0.001
R 896 336 896 304 0 0 40 10 0 0 0.5
d 896 336 896 368 2 1N4148
w 848 448 896 448 0
w 896 368 784 368 0
r 784 368 784 432 0 1000
t 752 448 784 448 0 1 -0.0010764116127125745 0.08970096770354627 100 default
g 784 464 784 480 0 0
r 752 448 704 448 0 5000
R 704 448 656 448 0 2 40 5 5 3.141592653589793 0.5
O 784 432 720 352 1 1
O 896 368 960 336 1 1
O 704 448 656 512 1 1
o 14 8 0 4098 10 0.1 0 2 14 3
o 15 8 0 4098 0.625 0.1 0 1
o 16 8 0 4098 10 0.1 0 1
o 1 8 0 4098 10 1.6 0 2 1 3
```
</details>

pfalstad, 2022-01-21:
Hmm, if you look at the console, it seems that the simulator is having a lot of trouble converging.  So maybe it's related to that.  If you take out the diode, replacing it with a wire, it works better.

colinhowell, 2022-01-21:
Yeah, it does look convergence-related.

Replacing the diode with a wire seems to work OK, but replacing it with a
resistor does not unless the resistance is very low (7 ohms or less).

closed 2022-07-11 by pfalstad