sharpie7/circuitjs1 #713: Feature- Request

Tonigau, 2022-01-23:
A  controlled switch would be a handy component. I tried using the Nch MOSFET(th=10, beta=200) with 32A 63V 50khz PWM & getting converge errors (doesn't error with default MOSFET parameters)
So a simple controlled switch would be handy, no config just 3 terminals .

cheers

pfalstad, 2022-01-23:
What about an Analog Switch?  Under Active Building Blocks

Tonigau, 2022-01-24:
Hello, I looked at the Analog Switch but could not get it to work.
I tried again & found it needs > 35V from WaveGen to work.
Thanks for the suggestion.
Don't get the Convergence Fail now with the switch node, but I don't get correct duty control of inductor current ?

[Buck2_Current_In_Asw.txt](https://github.com/sharpie7/circuitjs1/files/7921993/Buck2_Current_In_Asw.txt)
[Buck2_Current_In_MOSFET.txt](https://github.com/sharpie7/circuitjs1/files/7921994/Buck2_Current_In_MOSFET.txt)

[image]
ckt w- Analog Switch

[image]
ckt w- MOSFET

pfalstad, 2022-01-24:
I was going to ask why you have one side of the square wave generator connected to one of the switch terms.  Doesn't seem very useful or realistic.  Then I realized my example circuit does that!

What's the issue with the inductor current?

Tonigau, 2022-01-24:
_>> one side of the square wave generator connected to one of the switch terminals...
    >> What's the issue with the inductor current?_
Sorted, just forgot to turn on brain first... circuit below ok
.
[image]

closed 2022-01-30 by pfalstad