sharpie7/circuitjs1 #714: [BUG] Variable Capacitor

OmegaHaxors, 2022-01-24:
Attempting to simulate a Varicap by creating a normal capacitor and giving it a capacitance slider will result in unrealistic results.
In a real Varicap, adjusting the capacitance will cause the voltage to increase and decrease as you adjust the size of the capacitator.
However, in the simulator the voltage remains static, which means energy is being created from nothing.

Would it be possible to adjust the voltage within a capacitor to compensate for the change in capacitance when increased or decreased?

pfalstad, 2022-01-24:
See #487

OmegaHaxors, 2022-02-01:
Would it be possible to add a dedicated Varicap component which behaves like a real one?

pfalstad, 2022-02-01:
I was thinking we could add a flag to the capacitor to have it behave differently when changing the capacitance.

OmegaHaxors, 2022-02-01:
I would appreciate that a lot.

pfalstad, 2022-02-02:
Sure, if we just assume that changing the inductance will keep energy constant.

pfalstad, 2022-04-03:
Shouldn't it be the charge that remains constant when changing the capacitance, not the energy?

OmegaHaxors, 2026-03-20:
> Opened a PR to fix this: [pfalstad#298](https://github.com/pfalstad/circuitjs1/pull/298)
> 
> The fix tracks the previous capacitance value and adjusts the voltage across the capacitor in `startIteration()` to conserve charge when capacitance changes at runtime (e.g. via slider). When capacitance changes from C_old to C_new, the voltage is scaled by `C_old / C_new`, preserving Q = C*V.
> 
> This applies to all code paths that change capacitance: slider adjustments, edit dialog, and the programmatic `setCapacitance()` API. PolarCapacitorElm inherits the fix automatically since it extends CapacitorElm.

I think making it an option rather than replacing the original behavior would work a lot better. Not everyone is going to expect/want changes in the slider to cause unpredictable voltage/current spikes, but for those of us who want it, we can enable it in the settings.

Actually a more graceful way of implementing this is to have clone Capacitor/Inductor -> Varicap/Variac components that exhibit the new behavior, it would also allow for you to mix and match between the component types within the same simulation. This would also allow for stuff like moving the tap of a transformer to dynamically adjust the frequency of a radio to tune it.