sharpie7/circuitjs1 #715: Request- Show drawn power in osci

LawnchairLarry, 2022-01-24:
I can't show the drawn power from a device in the Osciloscope. not even from the powermeter xD

I was building an Powerampfifier for presentations and there is no way to compare the compare the power. I just can show the drawn power of the power source, thats it.

greetings and good Work so far! =)
I would spend you an coffee if there where a "spend him a coffee" butten somewhere around here :D

pfalstad, 2022-01-24:
Show a component in the scope, click the gear on the scope, and in the scope properties under “plots”, check “show power consumed”

LawnchairLarry, 2022-01-24:
Hmm i just figgured out that it works if i put the resistor on an extra osci for some reason. Hmm why is that?

LawnchairLarry, 2022-01-24:
Hey i think that is a wierd bug, i can add the Resistor to the osci first and then show tho power consuption. After adding the Power source to the same osci i cant push the button show power consuption anymore.

LawnchairLarry, 2022-01-24:
Nope, doesn't even work in browser version: https://tinyurl.com/ycgo8442
i just can show the power consuption of one device in an osci, after that the "Show power consumption button" doesn't a shit anymore. So i can't compare input and output power of that power amplifier =/

pfalstad, 2022-01-24:
that's weird all right...

LawnchairLarry, 2022-01-24:
So doctor "pfalsdtad" to the op pleace? We need to do an operation on the open sorce code =)

pfalstad, 2022-01-24:
Actually this example you sent has both the power and the voltage plotted in the same graph.  How'd you do that?  The green is voltage, the gray is power.

Obviously we should handle it better..  The power checkbox shouldn't just do nothing.

But here's how I view the power of any component: right click the component, view in scope, click on the scope gear, check "Show Power Consumed".  That doesn't work for you?

LawnchairLarry, 2022-01-25:
No, take a clother look: The one grey power plot is from the power sorurce. the other ones are from the resistor.

pfalstad, 2022-01-25:
Plot one, switch to Show Power, plot the other, switch to Show Power, combine:

https://tinyurl.com/y9qcw9gu

After you combine, the checkbox doesn't work.  I may just gray it out.  I don't know what that checkbox would do if you're already viewing the power of one device and the voltage of another.

LawnchairLarry, 2022-01-25:
> Android [4].

I see! Maybe you need to make a checkbox for eacht plot =/

Links:
------
[1] 
https://github.com/sharpie7/circuitjs1/issues/715#issuecomment-1021452013
[2] 
https://github.com/
[3] 
[4]

pfalstad, 2022-01-25:
yes, we really should.

LawnchairLarry, 2022-01-25:
And you should implement an "spend a coffee for dev" button ;)

Links:
------
[1] 
https://github.com/sharpie7/circuitjs1/issues/715#issuecomment-1021536877
[2] 
https://github.com/
[3] 
[4]