sharpie7/circuitjs1 #716: Feature- Math channel in scope

Tonigau, 2022-01-25:
.
Are there plans to add math channel plot to scope in future version.?
I am plotting Volt & Current, it would be good if I can also plot the power (ch1*ch2)in the same scope.
I can plot the power of the load resistor in another scope, but its not really V*I of the source & it would be nice in the same scope with I & V to export as vector image

[image]
.

pfalstad, 2022-01-25:
That was #86.  I closed it because you can use a VCVS to do math operations which you can then plot in a scope.  In this case, you can just use a wattmeter (under Outputs and Labels).  Also, any scopes can be combined.  (Right click on the scope, Combine)

Tonigau, 2022-01-25:
Using Circuit Simulator version 2.6.4js. (Win version)

I adding the watt meter. its working in scope  Tried combining scope - took me a while... (Scope can only be combined when docked.) ok
When I docked the watt meter watts plot now displays volts & tried check power consumed but stays at 0W , cant find how to un-combine.
Delete scope & start again now I cant get any watts displayed in single scope.
zzzzz...  I will give it a rest for now, try again in the morning.

The watt meter seems a bit ambiguous How to connect. It has 4 terminals  2 are current other 2 are Volt? (Nope)
Does it work with just 2 pins only for current & it assumes there is a Ground reference ?, needs also a Volt connect?
Sorry, I'm just struggling to get V I P displayed in my scope. (I dont know where to start with the VCVS yet)
I tried the help pdf for both. 

[PV_Panel3.txt](https://github.com/sharpie7/circuitjs1/files/7931026/PV_Panel3.txt)

I still think a math plot would be nice later down the track (its already there of sorts in the X-Y plot)

pfalstad, 2022-01-25:
Like this: https://tinyurl.com/ycnz2o2l

Tonigau, 2022-01-25:
Thanks heaps, Thats great.