sharpie7/circuitjs1 #717: Reuse circuit section

Tonigau, 2022-01-27:
Win offline version.
I found myself needing to copy a circuit section of another file, rather than re-drawing it all.
1. Tried running another instance of CircuitsJS but the clipboard is not shared.
2. Tried importing text of the circuit section but it removes existing circuit elements.
3. Workaround - what did work was to save or copy the circuit section text & then edit into the new circuit text file & file open.

Maybe an check option in the text import " Retain existing circuit" , or there is probably a way I haven't found.

pfalstad, 2022-01-27:
File->New Window
select all, copy, switch to other window, paste.

Tonigau, 2022-01-29:
Thanks,  "The File->New Window" copy Paste between windows works.
I will use this method.
_There must be some difference between when "a new instance is run" & "new window" (both show the same in task manager)

closed 2022-01-30 by pfalstad