sharpie7/circuitjs1 #722: New resource

ArmoredRodent, 2022-02-06:
Not sure where to put this.

I have a new hosting of the simulator with new circuits.  I've been using the simulator via tinyurl for years, so I figured it was time to give back to the community. 

Site: [https://wulflemm.com/war/circuitjs.html](https://wulflemm.com/war/circuitjs.html)

Under Applied directory, there are 30+ new and improved circuits.  Some are copies of existing circuits I plan to use and or modify.  Some are resources from around the net under tinyurl's (duly attributed) that deserve a home.  

I can zip up the new/modified versions and my directory structure, with a list of new/modifications, if you want to add them to the main library. 

Good sections on A/D, voltage regulators and timing.  I challanged myself to 30 circuits in 30 days.  There are a few other circuits that are not ready for distribution.  My goal was to give me a deeper level of understanding about how certain circuits worked.

[Buck Converter with PID PWM](https://wulflemm.com/war/circuitjs.html?startCircuit=../applied/switcher-buck+pwm+pid.txt)

[8-bit SAR ADC](https://wulflemm.com/war/circuitjs.html?startCircuit=../applied/adc-sar.txt)

Thank you for the excellent application!

To all, enjoy....

pfalstad, 2022-02-06:
A lot of good examples in there.  Thank you!

ArmoredRodent, 2022-05-04:
[circuits.zip](https://github.com/sharpie7/circuitjs1/files/8617298/circuits.zip)

Attached is a zip file of txt files.  Included is setupappliedlist.txt, which shows how I broke files into directories.  And circuits.pdf, which is setupappliedlist.txt with explanations of modifications to existing files.  Names for new files should be all that should be needed for an explanation.  Some examples come from the web, and I have attributed the source web location in the txt file.

Feel free to use what you feel adds value to the database.  Or not.  I'm fine either way.