sharpie7/circuitjs1 #723: Voltage not capped

danielkucera, 2022-02-06:
Is it me lacking understanding or is there something wrong?
Why is the voltage on 10k resistor going to 8V when there is a diode across it?
https://tinyurl.com/y958ogeq

pfalstad, 2022-02-06:
Yes that diode is acting strange.  It seems to have a series resistance of 12 ohms, but that isn't showing up when you try to edit the model.  I'm not sure how that happened.

pfalstad, 2022-02-06:
Looks like if you set the diode model on a diode to 1N5712 and then click "Create Simple Model" and then click OK, it will actually create a broken diode model with a series resistance of 12 ohms.

pfalstad, 2022-02-19:
To get this example to work you need to edit the broken diode model and then click "OK" without changing anything.

closed 2022-02-19 by pfalstad