sharpie7/circuitjs1 #728: Bug: Periodic voltage generators: Doesn't allow high frequencies

MrStark135, 2022-02-18:
Whenever I place a square wave generator, or a simple AC voltage generator, and try to increase the frequency above 25KHz I think, I get the pop-up box saying: "Adjust timestamp for higher frequencies?" with a yes and no buttons. But no matter which one I click, it does it's thing and sets the frequency to te desired value. However, after that I cannot change any parameters of ANY component in the circuit. It opens the pop-up box, eeverything seems fine, but I cannot change, for example resistance of a resistor. The buttons work, though. Like flip X, flip Y, swap E/C (for transistors) etc., but the text boxes for capacitance, resistance, etc. don't work. They look like a read-only text (I can select and copy it, but when I click the text box, it doesn't focus on it). So the whole circuit seems irresponsive after that. All I can do is watch the simulation. Press play, stop and reset, but the problem persists. Intrestingly enough, after a while, it "unlocks" itself from this behaviour, and I can edit the components again. But this seems random, not like because I pressed this and that. One more thing: THIS happens only after the "Adjust timestamp for higher frequencies". If I stay in the "low frequency range", the simulator is working fine. 
[image]

pfalstad, 2022-02-18:
Is this the offline version?  If so, I think this is #596.

MrStark135, 2022-02-19:
Yes, it's the offline version. Now that you mentioned #596, yes, it's the same. Sorry for duplicate, I checked only the titles of other bug reports, and I didn't find anything related to frequency changing; so I figured I should post it. Is it fixed? I have downloaded the offline version about two years ago.

MrStark135, 2022-02-19:
> Did you try Options > Other Options (at bottom)
>
> Under Edit Component adjust Time step size (s) to a lower value. 100n for
> 100ns. I believe this adjust the step size for the simulator.
>
> [image: image]
> <https://user-images.githubusercontent.com/16496324/154723993-eef66fda-e2cf-47a5-a6b2-cbfd196306e3.png>
>
> [image: image]
> <https://user-images.githubusercontent.com/16496324/154724053-47af1b93-4519-46d5-9413-51055278d911.png>

Yes, ikt works, but I was talking about the pop-up box that appears when
setting a high frequency for a square wave generator, for example. After
the pop-up, all text boxes lose focus, and I can't edit the components.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

pfalstad, 2022-02-19:
It's not fixed.  But if you click on another app window, and then switch back, things should work again.

MrStark135, 2022-02-21:
Ok, that'll do it. Thanks

closed 2022-02-21 by MrStark135