sharpie7/circuitjs1 #729: NPN&PNP behave funny

riddicc, 2022-02-27:
Hi!
It is about [this circuit](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcZYA4BsB2ATNzZsEBOVA9EAFgGYRsQFJ6BTAWjDACgB3EKyC8OkZ8B7Opx7ZIqcJnJ5xc8BwAutHLPIjNUEGz0wE6VBWxUTqDESpgEAljHRFM-XJHZVCn8mDggAzgAOAJYAxqwAJkwAZgCGAK4ANiocAEq8-Mh02qiMeZSMLrp50AgcAE600joEMmBKjL7wcBUZokTZmYTkjc0tAPJt4B28COIjEJAcAObg2DJSdfO02AJ5HABuQ9ra3cUFVfuGM9uZO1LFHIO7RqN0e5Oq9Er15EJ1DeB6TtBEf+gAgGYVD4Yx6MBEX5-UEUAGoVb4JBNEBROJJFLpd5ZEBY3L7CiFfIlMo8LFgIQ46piZSDMkjdC3CHiKCtOl0LF7Xp9VoIAg6XniPlc5rceh86lk8UnAVVGQy3BrFk8GWvZ4+BpXSkLW4M+SMlmzLHaI0XdZbI2ZDm3fIEw5E0ocaKUdDtOgqkaecClKCwbAcAAyIDxTJxrv2cUS-iYcxZgdhYYoFDqI0aIAjUZjUydwYmyxDnpQSBgfU4QA).
When I turn on a PNP rapidly, a seemingly unrelatee NPN goes mad...
Why is that?
Furthermore the PNP does not work as expected, because the left FET has high leakage...
Is that realistic?
Thx.
Bye.

pfalstad, 2022-02-27:
Looks like a convergence/roundoff error issue.  If you look at the console, the simulator is constantly having trouble converging even with no change in the circuit.  So no, it's not realistic.

Duckle29, 2022-03-28:
This circuit also has issues with this https://tinyurl.com/yaepqntf

pfalstad, 2022-03-28:
Yes..  To fix that one, you just need to add a capacitor to simulate the capacitance between the MOSFET gate and drain.

https://tinyurl.com/yd8d4pon

closed 2022-07-11 by pfalstad