sharpie7/circuitjs1 #737: Incorrect voltage with two voltage sources in one document

jbtls, 2022-03-15:
Hello, when I include two voltage sources in a document and build one series and one parallel circuit each, I get 5.6nV displayed at the second voltage source instead of 0V due to errors. If I remove one of the two sources, the error disappears. 

-> [Example Circuit](https://falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgpABZsKBTAWjDACgB3EFwqsYlNzC9wKPFE5CRYMeEKjx2ScIVzwAkCmXyZ4sDRqqlXfYd3gD6wVpPyxe+dgSDjFw-bcgnL5Ze9r-VxV-HipAyVCvZykqD0g2ACc1fkEVOIp4pNMrT3SwODYANxjNPHFIlKgqmli4KqoYBCKvNFLFVsqG2lquhugmrmwOjSG+WXjB1o9RqMFM2kIzWRpF2aqnaDxiSRn-FcNwybHxfaM2IA)

pfalstad, 2022-03-15:
This seems to be caused by roundoff error in the matrix.  Add explicit grounds to get rid of it.

https://tinyurl.com/y7fwhskv

jbtls, 2022-03-17:
Thx, so this is not a bug but a feature? ;)

pfalstad, 2022-03-17:
Well not a feature but not sure if it's a bug either.  The simulator is never going to be perfect at guessing where the ground is supposed to be.  I can probably fix the roundoff error issue though.

jbtls, 2022-03-18:
Ah, ok. THX anyway for this great software. I use it often with my students.

closed 2022-03-18 by jbtls