sharpie7/circuitjs1 #738: capacitors and grounding?

v-alisei, 2022-03-23:
Hello. I'm new to electronics. And have been confused with schematic attached. I would appreciate any help and explanations about what is going on.
[image]
[circuit-20220323-0300.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/8328575/circuit-20220323-0300.circuitjs.txt)

pfalstad, 2022-03-23:
looks right.  When you click that switch, all you're doing is changing which node is considered ground.  there's no current flow.

the voltage source and the resistor are not part of a complete circuit, and can be ignored (unless one of the other switches is clicked).  and the capacitor is not part of a complete circuit either, so there will not be any current through it regardless of that switch position.

v-alisei, 2022-03-23:
I see... you know, your program gives an intuition that when we have a charged capacitor, there is two options: either we have electron shortage on positive terminal, or surplus on negative, while in real life both of them are necessary and happening in the same time) Maybe there is a way to show this somehow. Anyways, your program is great tool for learning! Thank you!

closed 2022-03-23 by v-alisei