sharpie7/circuitjs1 #739: Transformation (turns) Ratio

SEVA77, 2022-03-26:
Question from my email:

> Transformation ratio = 10. This is a step-down transformer. Why is the voltage going up in your simulator? К= U1 / U2. 
The step-up transformer ratio must be 0...0.99 
> [image]

I repeated this:
```
$ 1 0.000005 89.03963197220334 47 5 30 5e-11
v 176 144 176 272 0 1 40 2 0 1.5707963267948966 0.5
w 352 224 352 272 0
T 272 192 352 192 0 4 10 0 0 0.9999
w 272 192 272 144 0
w 272 224 272 272 0
w 176 272 272 272 0
w 352 272 448 272 0
w 352 144 448 144 0
r 448 144 448 272 0 100
w 352 192 352 144 0
w 176 144 272 144 0
o 0 64 0 20483 5 6.4 0 2 0 3
o 8 64 0 20483 40 0.4 1 2 8 3
```

Transformation or turns ratio for me is defined as the ratio of primary to secondary voltage or turns but after reading some other sources I found other interpretations of this. In your example circuits ratio = 0.1 for a step-down transformer and  ratio = 10 for a step-up transformer.

Well, what is the Ratio?

pfalstad, 2022-03-26:
Ratio is turns in the secondary / turns in the primary.  I guess I used a different source, or screwed up.

When the ratio is displayed in the lower right, it is displayed in a different form that might make more sense.  For example, it says 1:10 for this transformer (which is acting as a step-up).  That's primary:secondary.

SEVA77, 2022-03-26:
In a [discussion about the definition of Turns Ratio](https://electronics.stackexchange.com/questions/409343/turns-ratio-and-transformation-ratio), I found the following:
> The IEEE standard does not define a "transformation ratio".
> However, the standard also says that the "turn ratio of a current transformer" is defined as the ratio of secondary to primary turns, while the "turn ratio of a voltage transformer" is defined as the ratio of primary to secondary turns.

pfalstad, 2022-03-26:
I could flip this and change the label so it’s unambiguous…

SEVA77, 2022-03-26:
It seems to me that the your transformer model can be both a current transformer and a voltage transformer. I think in the transformer model itself, you can simply make a choice of the type of transformer...

closed 2022-07-11 by pfalstad