sharpie7/circuitjs1 #745: Transistor Emittor-Collector Junction not breaking down

nithinmurali, 2022-03-29:
Currently, simulating circuits like this (LED blinker) doesn't work. Its because the p-n junction in the transistor is never breaking down (avalanche effect) even if high voltage is given across it. Can we make the transistor more realistic?

[image]

http://www.electronicshobbyprojects.com/basic-circuits/blinking-led-with-single-transistor.html

colinhowell, 2022-03-29:
I suspect it used a transistor because the designer specifically wanted to show off this sort of bipolar transistor breakdown. It doesn't surprise me much that the SPICE model doesn't support this breakdown, since it's not how bipolar transistors are normally used.

Where things really get weird is when someone designs a circuit like this with a reverse-biased bipolar transistor in breakdown, or just on the verge of it, because they're using the noisy avalanche-breakdown current as a noise source. A number of analog sound circuits for early arcade video games (and for some analog synthesizers) were built with noise generators like that. Others used higher-voltage "Zener" diodes (which are really avalanche diodes at these higher voltages) to get this effect. I'm not sure of the reason for the different approaches: was the character of the generated noise different, or was it a question of what parts were available and would behave as required? (This sort of behavior definitely would have been outside of the part manufacturer's design spec.) I'm sure that reliability and low cost were factors.

Anyway, just a bit of added background.

Noitarud, 2023-10-09:
I realised a new way to use a transistor too, the simulator was comfortable with it, the capacitor voltage rose gradually from zero, but in reality when i built it (built as reverse polarity), the capacitor rapidly charged to 1.7V then it was gradual. I suspect the other transistor in the circuit must be doing something undocumented, @pfalstad do the SPICE people accept such advice? I notice you have (on this issue) not said breakdown is an upstream issue, so the Spice simulation is not under centralised development?

The new transistor use was that i can use it as a deamplifier just as i can to amplify, by swapping the purpose of the Emitter and the Base, using the base as the output instead. Application was for a toaster timer, subsequent problems include capacitor impedace problems and needed to try a darlington on the trigger transistor (the Jfet could power the trigger but timer was 12s. still going).

Deamplifier circuit (I am using v1.5n still, replace semicolon with CR)
`$ 1 5.0E-4 0.551975421667673 50 5.0 50;R 368 160 400 160 0 0 40.0 10.0 0.0 0.0 0.5;t 288 224 368 224 0 1 -0.042643564338678885 0.6332109005735092 100.0;g 368 304 368 320 0;174 432 256 432 320 0 1000.0 0.9158000000000001 Current Adj;w 368 208 368 160 1;w 368 160 288 160 1;174 288 160 256 224 0 1000.0 0.5693 Load;O 288 224 224 224 1;i 368 240 368 304 0 0.01;w 256 192 288 160 0`
Toaster
`$ 1 5.0E-4 32.755850052045055 50 5.0 50;R 608 320 608 352 0 0 40.0 -24.0 0.0 0.0 0.5;t 400 256 416 256 0 1 -8.451270380423418 0.4236121876928678 100.0;t 416 176 560 176 0 -1 -0.5528068904393418 -0.6016016382744472 100.0;c 352 256 352 320 0 9.999999999999999E-6 15.548729619066577;g 448 112 464 112 0;x 296 211 376 217 0 24 Toaster;s 448 112 448 160 0 0 false;r 352 256 400 256 0 10000.0;r 448 320 448 272 0 900.0;w 352 320 448 320 0;r 432 256 416 176 0 100.0;w 560 160 448 160 0;162 608 272 608 320 1 2.1024259 1.0 0.0 0.0;w 448 320 560 320 0;r 560 192 560 320 0 10000.0;174 240 160 272 160 0 150000.0 0.9455000000000001 Heating;t 560 192 608 192 0 -1 23.673365144281924 -0.04879474783510536 100.0;w 560 160 608 160 0;w 608 160 608 176 0;w 560 320 608 320 0;r 608 272 608 208 0 100.0;r 240 240 240 160 0 150000.0;s 544 144 592 144 0 1 true;w 560 192 592 144 0;w 544 144 560 160 0;w 288 256 352 256 1;w 272 160 448 160 1;w 272 160 256 144 0;162 448 160 448 208 1 3.6 0.2 0.2 1.0;t 288 256 256 256 0 -1 15.548729619066577 -0.48562087772328244 100.0;w 352 320 256 320 1;w 256 320 256 272 0;w 528 256 560 256 0;162 448 208 448 240 1 3.6 0.2 0.2 1.0;162 448 240 448 272 1 3.6 0.2 0.2 1.0;j 112 256 160 256 1 -4.0;w 192 256 192 128 0;w 192 128 272 128 0;w 272 160 272 128 0;w 112 304 160 288 0;w 240 240 256 240 0;t 416 272 432 272 0 1 -8.250656583596 0.5389902006247276 100.0;r 416 240 448 160 0 100.0;w 432 288 448 272 0;o 3 64 0 33 10.0 9.765625E-5 0 -1;o 12 64 0 33 1.1692013098647223 0.37414441915671115 0 -1;o 10 64 0 289 0.018268770466636285 3.6537540933272573E-4 1 -1;o 7 64 0 33 1.4272476927059598E-4 9.134385233318143E-5 1 -1;o 3 64 0 35 17.498005798264096 8.543948143683641E-5 2 -1`

NickLarsenNZ, 2025-09-14:
I was looking for a way to simulate the "Esaki Oscillator" in Falstad and managed to stumble across it here (wanted to mention that here so that it comes up in searches).

> I wonder why this circuit doesn't just use a zener?

I also tried using a Zener Diode, and perhaps my values are just not correct enough to make the effect work, but my (little) understanding was that there were some characteristics of (some/many) NPN transistors that allowed the oscillation to occur.

I believe it is that the breakdown voltage decreases as the the current increases (after breakdown), so as the capacitor discharges, the voltage at the emitter goes below the new breakdown voltage and the NPN stops passing current, and the capacitor charges again. _I don't know if this is true, but it is my current mental model_.

Perhaps oscillation can be done with a Zener Diode, but I haven't found any information on how that can be achieved.