sharpie7/circuitjs1 #748: mouse drag is not working in this particular circuit for some reason

ericfont, 2022-03-31:
With the attached circuit ([with-diode-output-led-prtoection_circuit-20220331-0020.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/8390797/with-diode-output-led-prtoection_circuit-20220331-0020.circuitjs.txt)) I am unable to easily drag to select an area with my mouse.  I can drag individual elements.  But when I try to select an area, for some reason one of the corners of the rectangle is fixed to be the top center of the screen. Also clicking with the mouse in the menu doesn't seem to do any thing.  I have to reload circuitjs and start with a fresh or other circuit to use my mouse again.

ericfont, 2022-03-31:
there is also a undocked scope to the lower right of the red LED on the right. But for some reason that undocked scope is not visible but I know it is there cause if I move the cursor over it, I can see the cursor on the scope.

I don't know exaclty how I managed to get this circuit in this corrupted(?) state...[My previous save of an earlier version of the circuit](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgbArCBMB0AMinJdApgWgOwgMzQBYF4BGCXeLCMADgppBpJCgNxcxJICgAlEAjXggSBAgMEixIYcILDcsdrJmwI3AE54a40eNw6RATmgyRKeJoFDjpwcOhZTwkhe4Bzaw6deYNBrLcAO4s0AwkJr4Rzh6hDI6mEGF+ATLc0PBGcbbZCWbwxEhGxSWlZVmKiLgkWIi0uFh0YFkksJCqPCEYbKYZDBh5fWldPdriAz4G4pYARiDYSSIZ89DFvpYAHvNY+CBYWRgQoox2SyAAsgD2ACZoADYAOgDOlwBmzwAWl0HPAC4faGeMwAhndgQA7ADGaGuz2BAFdrgBLS7PSGXAC2aCeAC5uFtsLh2K5mBgGqZXOJxAxAEQEzzA8GeAAkAF7PcGXJFPQFPAAUACEAJTPV4aTHPAAO3zQGmedyR4Oxz3caF+Tzh11usKeCslH2B3L+qJeGiR7gVoOeOvc4MtfIAgsKnhDtb9jQqAG4yw1Pa22x58jCO2DpTLzEgMsYreRRlyYeIIIy4eS4IyiMBgEhTYltKAFHgZA4RhzwfqEEupEjxmCwaBgAzVNOrfYEIzkES5jrpEimQ6RoaHHswUtmUzAh4SmaxbpgKkR+YEWcSVKWTwz8TlhdLvKBLoMhhTeb74cBYJHwh4ZYYMAXgjzyx72-zw6RO9gYbn-RXm8bmMPz8wDGL69H+3Apgu84DhQG4+MIWAAPoRvB8CIbgRy1IhtYQMhsAkPBGB4dA8EIbgYHwOw3TztB8zUbgtBmDQiFgMhqHoShrTQNh+b4YR8GMaRXRGO+UG9COljQEOfYKIYg6mIeDjzOOk74keRg4BkpIMiWVJnPazxcs8tw6hoMJWma-oqYSWRiBREBUBI766KY-L6eqTzwuCRlIiZ2oclyPJus8JkYpcXqhjgSThJE1AuNFIAAGr6ncrzwTMSIGlYS6HmScmGHItSIGe27+DAPhhKe4E2NAF5DOVZgUExLFZmx8FcbhPHwURDUCTRuXjLs1H-llhjFaeWgEZG2UDXlAgFZY-CQdQIjCReKgxnm+RqLEVHLFRdH5GeOUnr1pUxAS5DvmhpKCMwaE6UO8XPA8ACOzxBk9ErPPylkQDGNRQN0NAtI0AhnI96qvU8enqp9TzfXMAycSI9kET48g4JsKy7KIvYRuEBCsLpADCaKXPCEryuC7iWdAFAiEc4Y2GQOkMPyZposCErApCSJuhoTw03TePhpGNTMNSIB0k8QQfEidw8lylxgr8CruM8AAiRMhudTM0ADi7hCVEuXKa5r+mZNp2ryjqSnc8JuX5PoCoKP32XRFGLgoGagwwnrejyfpW0GToU-b7Kck7QqC8wZBFnZy2nLSkoGtybkIsixqimgz1PNr2wuFmIhAyI84S1L-KXP8+nghK8JqiKJvPJcXMYrDLr6fXLx17XvzPEEcsBjMaDoliaJoOCvwyqZzpih52qtJAHqWfIzAZL21TqfAhMMIHAbWyHdsOxHPLO8v1V4CDZKUNo4uMCAfv8wH5lBzbTyh0f-nPKfCM+FwWRQBJd8mNrzEngOvcwhMhyay+ulHUVMyIHF2HVUaeBVBANUGA6oFhsG9HCidYs+DkhQESqCFKaUMoeiPEbZ8P4BD3nAAIWQuF8iqHUHuEqA5jxDHEmGDAQlTrzH4YNEQ1Y4BiAjI0BoSQfxAygAvPMuErBHREstFQrhkB8CoQwN82xlg6IUOAJhG0jGaOvBwkc2BljcJEOYJAaCVysO4E8eYEBIjZRirGMwrxQTcicS4yIA4PHWOEN4u4vjnHAToe+KSUSzDMFCeE-xIFhAxM3C4EACS0CHWPO4txeVskjkCQEsS3AADyAi6r4AcCVQIa4hHwHGEI5YgRxqLg3BY0Y1iICFVacNcYox5IsB6YI4SPg+HCTEiwVwSiCEDn4V06ZBIdjaMzEeHYAglxl1tmHaWstJ7PAAHLHxDHMAgjgRD7UzAociaQthnOpOITMUAaA4CcglJKZDYHPAAETxW8r8eEdoADis9PLfLRCZYEKtLjghUmcqA-DmjiDWFsp4ABlJErd5bPA9ErX4wIVSGSRB6JEtxZQvHeE8MEGhCXcmeoCgMJkdRPD5m5XkGIFSYqRCyHkUoggyiJb6JEXNoXghdncusEDDHvlcNou+5d2ZPCJpzI0nwlbanROCFlEJe5UrQF6AM7duRenBE3Ou4d-JwvOVwd8EBi5-x9pLZ4JDkqpS+U8SEEIgQB31D5IEABPZ4GJ4R3BVhTHkPMNCQnhLzAWWwsAyqwJvGVNhfwgEhPLL1Xdfg9xUkzNMw5iRGBZk6ll-doRN1NhaAMu8cTPGhhgL+wcnqQ2gPaPNq8wHGE7dvUt-d-hk11RCQNTxHZoDrXDN6TahQtueJjKqxwJKrxqI6qW0IJ5T21MCUF2oXWfINBOic8il5LIMAIBm15C4ExLVLX1IVwRIkhG5UU4ongMmZGyN+0pyWU2xJZHY7AjApLosIJMq7ngYstnvV+BldlQtcoZbE3kYSwCelDBDvpx66t1E8GYlcPhBTQO4EN264SeWrl6DQk9tSvBNhiAANECc1zo7gwrVtLXmBH3KeSQ360dxz-1EhLr2HRXBe1SwACoAhMhhl4o9s0930q3GEEIK3OnI08JlXJ8VQh5KCNjfwAR92Q88AAkgAci9MqWeEpp4GmeN0xkLwPgYm4NcARRJRm9A4SIA5dkuCxDyJ5jz9FAjucGBwsqMYKR+awAF-ggxotJZUHIYQxjHHjVmRY2ZsFbEaLXDl3G-YalpAK5GMQ-QCFtIOplyaMkquSBcG4SEKwyoWMGLlqsfDVDu32HawQEBqCNAiMoQ6gx2vRZKWuDrvYIsrmnHkCrrW7BLkCC1iYfVlvLjiZgA4BQxBUBeasNgkAmCLH-BtqMAwYzyW4C1vp4AqSNYELALAnA5KwHkBJTIxR-BCTTCu1cj3tubNW2kFrAzFijHkClwopAMHw4KIgHg7n7BSCpPm3oIBbjeJDb8NzLBpB0QYL9J78RsdoFx6G7glw8ByDiSdkgB42g9hMJmEwNB6zkRuUj5HZhLoGIxnsVBqQDzcDPa4kAAAxZgajwwXGBBsZ1eKCVZIl1kGXqDXB5ZScwKXJl6Xj0hP67gQA) (before I made some changes and added an LED), seems to work just fine.

pfalstad, 2022-03-31:
I just fixed a few bugs related to undocked scopes, but they haven't been released yet.  Deleting an undocked scope would leave an invisible scope lying around, like this.  Also when creating an undocked scope, it wouldn't work until you resized it.

ArmoredRodent, 2022-04-06:
The file breaks the simulator.  Nothing works.  None of the menus do anything.

Paul is correct.  The circuit shows 4 undocked scopes.  The text file has 6 undocked scopes (Code 403).  

I deleted all and the circuit works.  A bit of trial and error will fix your problem.

pfalstad, 2022-04-07:
this is fixed in the latest version.

closed 2022-07-11 by pfalstad