sharpie7/circuitjs1 #749: Bug - LM317 Low Current Limit

themetalmuncher, 2022-03-31:
Not completely sure if this is a bug or intended behaviour, but the LM317 model seems to have an output current limit of about 111mA. 

Example circuit: https://tinyurl.com/ybh9abtn

There are some variants that do have a listed output current of about 100mA, e.g. TI LM317L, but most of the standard LM317 list an output current of 1.5A.

It was not immediately clear to me what was going on in my circuit when I ran into this, perhaps if the existing model can't be changed it could be renamed LM317L to make it more clear its not a standard LM317 variant and not to expect 1.5A out of it. But I think it would be nice if the sim did include a normal 1.5A output version.

pfalstad, 2022-04-01:
The LM317 model is derived from a spice model I got from somewhere.  When I try that model in spice I can get the output up to around 700mA if I change the test circuit a bit.

I found some bugs in the model conversion and once those are fixed, the output is up to about 190mA.  I assume there are some other bugs in the conversion so I'll keep looking.

pfalstad, 2022-04-02:
I found some other bugs in the conversion so now I was able to get the LM317 model to act the same as the spice model.  Not sure where to find a better spice model.

themetalmuncher, 2022-04-07:
Thanks for looking at this, certainly an improvement.

I had a bit of a look around for models, I can get about 1.2A out of this guy's model in LTSPICE: https://www.electro-tech-online.com/threads/lm317-spice-model.18270/#post-1185720

Setting R26 even lower nets even more current, but not sure what other effects that might have.

MikePryadko, 2022-11-01:
Sorry me for dummy question, but: why do you use SPICE in this simulator? This simulator is simple and visual! If somebody needs precision and "scientific" he will use special software, complex and incomprehensible. I think for LM317 should be enough this:
[image]
(picture is from **Texas Instruments**'s datasheet)

pfalstad, 2022-11-03:
what's in the "over temp & over current protection" box?

MikePryadko, 2022-11-04:
is this a rhetorical question? does any element in this simulator have "over temp" or "over current" protection?
why do you need a linear stabilizer like LM317 in this simulator at all? because it has non-obvious feedback circuit and for complex cases (like CV & CC same time) you'd like to "see" how it works before implementing it in PCB! so do you really care about "over temp & over current protection"? (last question is rhetorical)