sharpie7/circuitjs1 #750: How to export current or resistance using Data export?

rrcmks, 2022-04-02:
The circuit simulator is an amazing program.
I love it many features. Thanks to the contributors.
I am exporting Voltage data through Data Export in Output options.
But I wish to export Current as well as Resistance by the same way.
I tried a lot of ways but I can't succeed. 
Please help to sort out this.
Thanks in advance
M K Saba
https://www.enote.page

pfalstad, 2022-04-02:
you can convert current to voltage using a CCVS: https://tinyurl.com/yde53ccj
then you can export it.

using that, you can also use a VCVS to measure the voltage and calculate the resistance:
https://tinyurl.com/ydh23v39

closed 2022-07-11 by pfalstad

rrcmks, 2022-07-11:
My previous request
The circuit simulator is an amazing program.
I love it many features. Thanks to the contributors.
I am exporting Voltage data through Data Export in Output options.
But I wish to export Current as well as Resistance by the same way.
I tried a lot of ways but I can't succeed.
Please help to sort out this.
Thanks in advance
M K Saba

Hope you are fine.
Thanks for your support.
As quoted....
Still I am unable to export Current data by data export it only export V.

I used these two methods, mentioned by you.

https://tinyurl.com/ydh23v39
https://tinyurl.com/yde53ccj

Still it does not work. It only exports V at all points.

Please provide a solution for this.