sharpie7/circuitjs1 #751: Creating subcircuits Error  "Some nodes are unconnected!"

Jara-da-Cimrman, 2022-04-02:
Error: There is a problem of creating more complex subcircuits (eg O.A. 339, 741, 431, 317, 723 ...) in the last few versions. You always get the error message "_Some nodes are unconnected_!" In online version 2.4.2, these subcircuits can be created without any problems. I can provide schematics of these circuits to verify the problem.

pfalstad, 2022-04-02:
hi, could you give me an example?

Jara-da-Cimrman, 2022-04-06:
Hi, sorry for the lagging
some schematics are in the annex
in the latest version, an error message is displayed when creating the subcircuit
in the online version 2.4.2 (https://lushprojects.com/circuitjs/circuitjs.html) creation the same subcircuit is functional

[MAA723.txt](https://github.com/sharpie7/circuitjs1/files/8433246/MAA723.txt)
[LM386.txt](https://github.com/sharpie7/circuitjs1/files/8433243/LM386.txt)
[LM723.txt](https://github.com/sharpie7/circuitjs1/files/8433232/LM723.txt)

Jara-da-Cimrman, 2022-04-10:
Hi, perhaps another useful note:
not all relatively complicated circuits cause this error. (attached) The error also sometimes does not occure if you randomly select only a part of the problematic circuit with at least some valid nodes. (Tested on the circuit MAA741).

[LM339.txt](https://github.com/sharpie7/circuitjs1/files/8457959/LM339.txt)
[TL431.txt](https://github.com/sharpie7/circuitjs1/files/8457960/TL431.txt)

closed 2022-07-11 by pfalstad