sharpie7/circuitjs1 #753: unable to change minimum slider value for a resistor

ericfont, 2022-04-04:
In [this circuit](https://github.com/sharpie7/circuitjs1/files/8413504/circuit-20220404-1833.circuitjs.txt), I am unable to change this 7k feedback resistor's minimum slider value:

[image]

If I try to change it's minimum slider value to 1k instead of 1, then nothing happens.  Note that this resistor slider is shared by that 7.9k resistor too, which does have 1k as its minimum slider value.

ericfont, 2022-04-04:
(on version 2.6.9js)

pfalstad, 2022-04-04:
workaround: set the "Share Slider" popup to "New Slider", set the min/max value, change the "Share Slider" popup back to "bottom-resistance"

closed 2022-07-11 by pfalstad