sharpie7/circuitjs1 #756: Sometimes the sine wave is inverted, but no reason.

lkz0ne, 2022-04-09:
Hi, i'm rewriting this on the right section because ive posted in Pull requests before.

I showed the falstad simulator to my teacher and we started to use the simulator in the class.
When we did to simulate SCR, the wave had output inverted by no reason.

Today I was testing somethings, and when i build a simple rectifier bridge with a tapped transformer, sometimes the output wave is inverted, sometimes not inverted.

Why it occurs? How to solve the problem?

Here's is my circuit example

The output waves are from both resistors.

file:///home/lkz/Downloads/circuitjs1/resources/app/war/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKADdwU9wNDPvmKKOBAAWWsKowELMCX7gwQtAKXDRIhjDgEECHLkjY8ozGA1hoGcygR4MxOXgSiwxRuW3pIaBGH3YZIS8eHh8MKQsACYgKorKofGxIFF0AGYAhgCuADYALtGx+Dx8XAK8yamZuQUA5kUCohpxclJQLADusYlgFWVJkCz12AhCgiAjY1ySLABOE6NJ-eNUYPAsAPYg5IQaVOKk1F6EtsJC5NgsQA

lkz0ne, 2022-04-09:
Hi Falstad,

I figured out what this bug is about.

It is about the resistor polarity (of course a bug, i think, because resistors have not polarity). When we invert the polarity of the Load resistor, the sine wave get upside down.

Can you fix it? Thanks.

pfalstad, 2022-04-09:
I think you just need to be consistent when creating resistors.  Start with the "high voltage" terminal first, then drag it to the "low voltage" terminal.

Or if you create it the wrong way, right click the resistor and select "Swap Terminals".

The only other way to fix this would be to show only the absolute value of the voltage/current when displaying a resistor in the scope.  This would break existing circuits and so I don't want to do that.

lkz0ne, 2022-04-09:
Ok. I am sorry.

closed 2022-04-10 by lkz0ne