sharpie7/circuitjs1 #759: Circuit for erroe search in a circuits for education

ClauKam, 2022-04-12:
For educational training it would bee a nice future, whenn you could made circuits with an error.
Somethin like that you have a transistor circuit and you could made failer simulation with some parts like.
R & C = with short circuit or interruption (open R like MOHM)
Transistor with short or open over Collector to Emiter and Basis to Emitter.
Then you have the posibbility to made failer search in a circuits.
Failer search is an important thing for people in electronic. If you have the posibility to define that a part will have a fault, like a shortening oder an openning part and you could it simulate with the correct voltage at the meserment point, you have a big advantic for the future.
Not all parts made sense, it could bee the most parts witch made often failer in a circuit.

pfalstad, 2022-04-13:
I would use subcircuits for this.

http://www.falstad.com/circuit/subcircuits.html

You can create a "black box" device which might be a working transistor, but it might have a fault.

ClauKam, 2022-04-14:
Thank you.
I will try it.
Thanks in advanced.

closed 2024-12-14 by pfalstad