sharpie7/circuitjs1 #761: Random crashes

ChromicQuanta, 2022-04-18:
It randomly freezes when I use it, probably happens more when I undo something.

pfalstad, 2022-04-18:
Is there anything in the console when that happens?

ChromicQuanta, 2022-04-18:
> circuitjs1-0.js:2598 Uncaught TypeError: Cannot read properties of undefined (reading '0')
> at Tkb.Ukb [as _e] (circuitjs1-0.js:2598)
> at Tkb.iV [as Qf] (circuitjs1-0.js:2595)
> at u2 (circuitjs1-0.js:2514)
> at b1 (circuitjs1-0.js:2493)
> at M3 (circuitjs1-0.js:2584)
> at L4.M4 [as ce] (circuitjs1-0.js:2596)
> at oh (circuitjs1-0.js:275)
> at Zg (circuitjs1-0.js:891)
> at ah (circuitjs1-0.js:1364)
> at eval (circuitjs1-0.js:1945)
> at d (circuitjs1-0.js:1905)

pfalstad, 2022-04-18:
Offline or web?  What version?

ChromicQuanta, 2022-04-18:
http://falstad.com/circuit/

ChromicQuanta, 2022-04-18:
Online I think

pfalstad, 2022-04-18:
could you post the circuit?

File->Export As Link
copy to clipboard
paste it here

closed 2022-07-11 by pfalstad