sharpie7/circuitjs1 #764: VCVS 5/a errors

Qwerty135246, 2022-04-24:
When I try to make a VCVS that has the function a*(5/a), it comes up with the error: NaN/infinite matrix! Link: https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAQBYKLIKBTAWjDAChLJCRK87tC83OCGx5qKcFFhIwALggS6dVgGMQKFNWz91mkQKhT4cCEgSqKKQdq4Ir+wUtjGIS1gBtwxCf0FhvIjp0rlCsAO5ePjoaWkHhkQ66WgaQrChgEigIXDYikLFcwSAAhvFZOdF6uakRMYl1lGCFZXqNOXbcTaER2B1tIh2+oQDmAxL9vePYiqEATty8gbb21cg1C3wGdtYprPOTS0J8QeDxHJvWg7vpEggYFVyEVScQxQBUABQIAPTFAJSsUaEDDjLrAqYzdbgxLQ-pQkGdZaCOHxbaIyw7RzxaG5EHHZoRHEGPGJVIABxAOJ0RL8hjYETuDwxh1SAHtJNNDAAPDLEMCQABG6ickCyhgkEGwrHZPhmPJQfMF6mgSBgcGwGRk6m1Upl2BEct5-KFavg9wV2GIlAQGrFMAUpyAA

pfalstad, 2022-04-24:
what's a*(5/a)?  that's just 5 isn't it?

the expression a*(5/a) is undefined if a is 0..  that's why you're getting that error.  the expression should not be infinite at any point, especially not at 0.

Qwerty135246, 2022-04-25:
Thank you! How can I improve, I want to amplify the sine wave with an amplification peak of 5V

pfalstad, 2022-04-25:
I'm confused about why you want to do that.  I could come up with something, but it wouldn't be practical circuit so there wouldn't be much point.  Are you trying to emulate a real device?

Qwerty135246, 2022-04-26:
I’m trying to create some form of transmitter/receiver

closed 2023-01-09 by pfalstad