sharpie7/circuitjs1 #765: The variable 'goodIteration' in runCircuit is always true, it will never be set to false.

Mervill, 2022-04-24:
If I had to guess, I'd say that it's supposed to be set to false when `runCircuit` false to converge:

```
if (subiter == subiterMax) {

    // convergence failed
    goodIterations = 0;
    goodIteration = false; // I think this line needs to be added.
    ...
```

pfalstad, 2022-04-24:
I don't remember but I would guess goodIteration should just be removed.  It seems to be dead code.  I can't think of a reason why we need it.

Mervill, 2022-04-25:
Just that value on it's own would be removed, correct? There's also an integer `goodIterations` (plural) that I think is involved in the scaling timestep i.e.

```java
if (goodIterations >= 3 && timeStep < maxTimeStep && goodIteration) {
	// things are going well, double the time step
	timeStep = Math.min(timeStep * 2, maxTimeStep);
	console("timestep up = " + timeStep + " at " + t);
	stampCircuit();
	goodIterations = 0;
}
```

So we'd keep all that, but just ax the Boolean, correct?

pfalstad, 2022-04-25:
yes

closed 2022-04-25 by Mervill